aristanetworks/sonic #10: Port ordering issue on Arista-7260-CX3 issue?

yxieca, 2017-08-16:
While bringing up Arista-7260-CX3 switch with SONiC image, we wired cable to front panel port 1 and it reached link up. However, when we look at the portstat output, the internal tool shows that port ce16 reached link up.

      ce16( 34)  up     4  100G  FD   SW  No   Forward  TX RX   None    F  CAUI4  9238    No      

Please let us know if this is expected?

Can we make internal port number match with the front panel port number?

lguohan, 2017-08-18:
need to update the config.bcm to make the bcm port ordering to match the front panel port ordering.

Staphylo, 2017-08-23:
This problem might also be related to the issue #8.
We are working on providing a fix for this one, so it might solve your problem.

Like lguohan mentioned it's also possible to change the logical port number by changing the mapping logical to physical in the bcm config.

yxieca, 2017-08-23:
Thanks Samuel,

I will be more than happy to test the fix!

Regards,
Ying

yxieca, 2017-08-25:
this issue is not addressed with sfp util change.

For instance, with 108x50G+10x100G configuraiton:

When Ethernet0 (front panel port 1) came up, xe28 becomes 'up' in "bcmcmd ps" output.
The mapping from front panel port number to xe or ce number seems to be shuffled.

For the purpose of helping debug in the future. we would like to see front panel port 1 maps to xe1 or ce1 depending on the speed, and so on.

Staphylo, 2017-11-21:
Closing the issue as the internals of the SDK prevents us from doing so just by editing the configuration.

closed 2017-11-21 by Staphylo