aristanetworks/sonic #13: tg3 is not ready till very late in the boot process

lguohan, 2017-08-24:
Aug 24 15:43:50.792315 str-a7050-acs-1 INFO systemd[1]: Starting LSB: Raise network interfaces....
Aug 24 15:43:50.824784 str-a7050-acs-1 WARNING kernel: [   23.231266] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:51.808794 str-a7050-acs-1 WARNING kernel: [   24.215258] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:52.792794 str-a7050-acs-1 WARNING kernel: [   25.199255] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:53.776797 str-a7050-acs-1 WARNING kernel: [   26.183260] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:54.760774 str-a7050-acs-1 WARNING kernel: [   27.167254] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:55.744767 str-a7050-acs-1 WARNING kernel: [   28.151251] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:56.728772 str-a7050-acs-1 WARNING kernel: [   29.135254] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:57.712768 str-a7050-acs-1 WARNING kernel: [   30.119251] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:58.696774 str-a7050-acs-1 WARNING kernel: [   31.103250] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:43:59.680768 str-a7050-acs-1 WARNING kernel: [   32.087249] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:00.664777 str-a7050-acs-1 WARNING kernel: [   33.071249] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:02.108333 str-a7050-acs-1 INFO rsyslogd: [origin software="rsyslogd" swVersion="8.4.2" x-pid="681" x-info="http://www.rsyslog.com"] rsyslogd was HUPed
Aug 24 15:44:25.060796 str-a7050-acs-1 WARNING kernel: [   57.467242] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:25.132771 str-a7050-acs-1 WARNING kernel: [   57.539239] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:25.700769 str-a7050-acs-1 WARNING kernel: [   58.107236] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:25.772767 str-a7050-acs-1 WARNING kernel: [   58.179235] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 1"
Aug 24 15:44:58.868799 str-a7050-acs-1 WARNING kernel: [   91.275230] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.292770 str-a7050-acs-1 WARNING kernel: [   91.699220] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.364768 str-a7050-acs-1 WARNING kernel: [   91.771222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.436768 str-a7050-acs-1 WARNING kernel: [   91.843225] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.508767 str-a7050-acs-1 WARNING kernel: [   91.915222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.580766 str-a7050-acs-1 WARNING kernel: [   91.987222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:44:59.652766 str-a7050-acs-1 WARNING kernel: [   92.059222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:00.028767 str-a7050-acs-1 WARNING kernel: [   92.435222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:00.852771 str-a7050-acs-1 WARNING kernel: [   93.259219] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:01.836787 str-a7050-acs-1 WARNING kernel: [   94.243222] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:02.820768 str-a7050-acs-1 WARNING kernel: [   95.227210] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:03.804772 str-a7050-acs-1 WARNING kernel: [   96.211230] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:04.788779 str-a7050-acs-1 WARNING kernel: [   97.195219] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:05.772768 str-a7050-acs-1 WARNING kernel: [   98.179218] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:06.756772 str-a7050-acs-1 WARNING kernel: [   99.163218] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:07.740796 str-a7050-acs-1 WARNING kernel: [  100.147230] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:08.724789 str-a7050-acs-1 WARNING kernel: [  101.131215] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:09.708773 str-a7050-acs-1 WARNING kernel: [  102.115216] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:10.692773 str-a7050-acs-1 WARNING kernel: [  103.099220] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:11.676768 str-a7050-acs-1 WARNING kernel: [  104.083216] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:36.052797 str-a7050-acs-1 WARNING kernel: [  128.459214] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:36.124771 str-a7050-acs-1 WARNING kernel: [  128.531204] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:36.692773 str-a7050-acs-1 WARNING kernel: [  129.099205] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:36.764766 str-a7050-acs-1 WARNING kernel: [  129.171203] sonic: smbus access failed address=0x4e on adapter="SCD SMBus master 0 bus 5"
Aug 24 15:45:37.308794 str-a7050-acs-1 DEBUG kernel: [  129.715257] tg3 0000:04:00.0: irq 77 for MSI/MSI-X
Aug 24 15:45:37.308831 str-a7050-acs-1 DEBUG kernel: [  129.715274] tg3 0000:04:00.0: irq 78 for MSI/MSI-X
Aug 24 15:45:37.308836 str-a7050-acs-1 DEBUG kernel: [  129.715285] tg3 0000:04:00.0: irq 79 for MSI/MSI-X
Aug 24 15:45:37.308840 str-a7050-acs-1 DEBUG kernel: [  129.715296] tg3 0000:04:00.0: irq 80 for MSI/MSI-X
Aug 24 15:45:37.308995 str-a7050-acs-1 DEBUG kernel: [  129.715307] tg3 0000:04:00.0: irq 81 for MSI/MSI-X
Aug 24 15:45:37.352809 str-a7050-acs-1 INFO kernel: [  129.759785] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

lguohan, 2017-08-24:
Aug 24 16:00:55.823443 str-a7050-acs-1 INFO systemd[1]: Starting LSB: Raise network interfaces....
Aug 24 16:02:43.431663 str-a7050-acs-1 INFO systemd[1]: Started LSB: Raise network interfaces..

it takes two minutes to raise eth0. other services all depend on this one.

Staphylo, 2017-11-03:
Fixed with #16

closed 2017-11-03 by Staphylo