aristanetworks/sonic #16: 7060CX-32S eth0 interface coming up after ~2 minutes

prsunny, 2017-10-23:
eth0 interface is found to be coming up only after ~2 minutes. 

Tested with the following version of Sonic image: 
admin@str-7060cx-acs-10:~$ show version 
SONiC Software Version: SONiC.HEAD.287-2a56479

Please find the timelines below: (Note: T2 is 1 minute 53 second AFTER T1)

Time                                         Event
=======                               ====
T0  0  sec                                  Executed reboot
T1   47 sec                                 Login prompt
T2   1 min 53 second                 eth0 is UP

prsunny, 2017-10-24:
In the case of image upgrade, eth0 is coming up after ~1 minute 35 sec after getting the login prompt. 

admin@str-7060cx-acs-10:~$ show version
SONiC Software Version: SONiC.HEAD.295-c74007b
Distribution: Debian 8.9
Kernel: 3.16.0-4-amd64

Staphylo, 2017-10-25:
There are two reasons why this could happen.

The first one is if there is no DHCP server available. In this case the networking service would hang until dhclient timeout (1min). Given that the default minigraph for Arista platform doesn't provide a `<ManagementIPInterfaces>` section, dhclient is invoked twice during the first boot which is aligned with the timing.

The second possibility which I think is the one hitting here is due to the `udevadm settle` command in the networking service `/etc/init.d/networking`
Given that the pmbus driver take a while to be loaded we moved their initialization in the background. However loading new drivers continue to generate udev events which is what `udevadm settle` is waiting until it ends.

Few solution from there would be
- Avoiding the `udevadm settle` on Arista platforms
- Adding a timeout to the command
- Finding a way to avoid for networking unrelated events
- Replacing the command by something like `udevadm settle --exit-if-exists /sys/class/net/eth0`

I'm currently investigating the 3rd solution as it seems to be the less invasive but I'm not totally sure about the feasibility.

closed 2017-11-03 by prsunny