aristanetworks/sonic #17: show presense does not show correct mapping

lguohan, 2017-10-27:
Ethernet132 - Ethernet256 has qsfp transceiver, should be present.

admin@str-7260cx3-acs-1:~$ sudo sfputil show presence
Port         Presence
-----------  -----------
Ethernet0    Present
Ethernet2    Present
Ethernet4    Present
Ethernet6    Present
Ethernet8    Present
Ethernet10   Present
Ethernet12   Present
Ethernet14   Present
Ethernet16   Present
Ethernet18   Present
Ethernet20   Present
Ethernet22   Present
Ethernet24   Present
Ethernet26   Present
Ethernet28   Present
Ethernet30   Present
Ethernet32   Present
Ethernet34   Present
Ethernet36   Present
Ethernet38   Present
Ethernet40   Present
Ethernet42   Present
Ethernet44   Present
Ethernet46   Present
Ethernet48   Present
Ethernet50   Present
Ethernet52   Present
Ethernet54   Present
Ethernet56   Present
Ethernet58   Present
Ethernet60   Present
Ethernet62   Present
Ethernet64   Present
Ethernet66   Present
Ethernet68   Present
Ethernet72   Present
Ethernet74   Present
Ethernet76   Present
Ethernet80   Present
Ethernet82   Present
Ethernet84   Present
Ethernet86   Present
Ethernet88   Present
Ethernet90   Present
Ethernet92   Present
Ethernet94   Present
Ethernet96   Present
Ethernet98   Present
Ethernet100  Present
Ethernet102  Present
Ethernet104  Present
Ethernet106  Present
Ethernet108  Present
Ethernet110  Present
Ethernet112  Present
Ethernet114  Present
Ethernet116  Present
Ethernet118  Present
Ethernet120  Present
Ethernet122  Present
Ethernet124  Present
Ethernet126  Present
Ethernet128  Present
Ethernet130  Present
Ethernet132  Not present
Ethernet134  Not present
Ethernet136  Not present
Ethernet138  Not present
Ethernet140  Not present
Ethernet142  Not present
Ethernet144  Not present
Ethernet146  Not present
Ethernet148  Not present
Ethernet150  Not present
Ethernet152  Not present
Ethernet154  Not present
Ethernet156  Not present
Ethernet158  Not present
Ethernet160  Not present
Ethernet162  Not present
Ethernet164  Not present
Ethernet166  Not present
Ethernet168  Not present
Ethernet170  Not present
Ethernet172  Not present
Ethernet174  Not present
Ethernet176  Not present
Ethernet180  Not present
Ethernet184  Not present
Ethernet188  Not present
Ethernet192  Not present
Ethernet196  Not present
Ethernet200  Not present
Ethernet204  Not present
Ethernet208  Not present
Ethernet210  Not present
Ethernet212  Not present
Ethernet214  Not present
Ethernet216  Not present
Ethernet218  Not present
Ethernet220  Not present
Ethernet222  Not present
Ethernet224  Not present
Ethernet226  Not present
Ethernet228  Not present
Ethernet230  Not present
Ethernet232  Not present
Ethernet234  Not present
Ethernet236  Not present
Ethernet238  Not present
Ethernet240  Not present
Ethernet242  Not present
Ethernet244  Not present
Ethernet246  Not present
Ethernet248  Not present
Ethernet250  Not present
Ethernet252  Not present
Ethernet254  Not present
Ethernet256  Not present
Ethernet260  Not present

Staphylo, 2017-10-27:
We use the "index" column of the port_config.ini to know what is the physical front panel port associated to the interface name. In the case of https://github.com/Azure/sonic-buildimage/blob/master/device/arista/x86_64-arista_7260cx3_64/Arista-7260CX3-D108C8/port_config.ini the port column is incremented for every interface which breaks this logic.
Alternatively https://github.com/Azure/sonic-buildimage/blob/master/device/arista/x86_64-arista_7050_qx32/Arista-7050-Q16S64/port_config.ini contains the information we expect.

If the "index" is not supposed to be used like we assumed, we can change the logic of our sfputil plugin to use the alias column.
If our assumption was right the port_config.ini will have to fixed.

Let me know how we should go forward, we need to have a reliable way to convert linux kernel interface into front panel port in the plugin.

lguohan, 2017-10-27:
yes the index is supposed to be the front panel port index.

Staphylo, 2017-11-02:
@lguohan should we update the port_config.ini on Azure/sonic-buildimage or is it already being taken care of?

Staphylo, 2017-11-03:
Issue addressed on Azure/sonic-buildimage

closed 2017-11-03 by Staphylo