aristanetworks/sonic #30: Port breakout and changed SFP port order

bluecmd, 2021-04-15:
Good day!

I have modified SONiC to introduce [Dynamic Port Breakout](https://github.com/Azure/SONiC/blob/master/doc/dynamic-port-breakout/sonic-dynamic-port-breakout-HLD.md) for my Arista 7050 QX32 inspired by the setup on EOS for this model - i.e. the first 24 ports can be configured as 4x10G, the last 8 ports are fixed 40G.

You can see my BCM file [here](https://github.com/bluecmd/sonic-buildimage/blob/arista-7050-qx32-breakout/device/arista/x86_64-arista_7050_qx32/Arista-7050-QX32/td2-a7050-qx32-32x40G.config.bcm).

It seems that SFP transceiver is dependent on the ASIC port ordering. You can see that quite clearly in this interface status display. Note how Ethernet64 and Ethernet68 are ports with QSFP transceivers present, but due to changed ASIC port ordering it seems they are now confused and mapped to Ethernet16 (and one transceiver is gone). 
```
admin@localhost:~$ show int status
  Interface            Lanes    Speed    MTU    FEC         Alias    Vlan    Oper    Admin            Type    Asym PFC
-----------  ---------------  -------  -----  -----  ------------  ------  ------  -------  --------------  ----------
  Ethernet0  125,126,127,128      40G   9100    N/A   Ethernet1/1  routed    down       up             N/A         N/A
  Ethernet4  121,122,123,124      40G   9100    N/A   Ethernet2/1  routed    down       up             N/A         N/A
  Ethernet8      13,14,15,16      40G   9100    N/A   Ethernet3/1  routed    down       up             N/A         N/A
 Ethernet12       9,10,11,12      40G   9100    N/A   Ethernet4/1  routed    down       up             N/A         N/A
 Ethernet16      17,18,19,20      40G   9100    N/A   Ethernet5/1  routed    down       up  QSFP+ or later         N/A
 Ethernet20      21,22,23,24      40G   9100    N/A   Ethernet6/1  routed    down       up             N/A         N/A
 Ethernet24      25,26,27,28      40G   9100    N/A   Ethernet7/1  routed    down       up             N/A         N/A
 Ethernet28      29,30,31,32      40G   9100    N/A   Ethernet8/1  routed    down       up             N/A         N/A
 Ethernet32      37,38,39,40      40G   9100    N/A   Ethernet9/1  routed    down       up             N/A         N/A
 Ethernet36      33,34,35,36      40G   9100    N/A  Ethernet10/1  routed    down       up             N/A         N/A
 Ethernet40      45,46,47,48      40G   9100    N/A  Ethernet11/1  routed    down       up             N/A         N/A
 Ethernet44      41,42,43,44      40G   9100    N/A  Ethernet12/1  routed    down       up             N/A         N/A
 Ethernet48      53,54,55,56      40G   9100    N/A  Ethernet13/1  routed    down       up             N/A         N/A
 Ethernet52      49,50,51,52      40G   9100    N/A  Ethernet14/1  routed    down       up             N/A         N/A
 Ethernet56      69,70,71,72      40G   9100    N/A  Ethernet15/1  routed    down       up             N/A         N/A
 Ethernet60      65,66,67,68      40G   9100    N/A  Ethernet16/1  routed    down       up             N/A         N/A
 Ethernet64               77      10G   9100    N/A  Ethernet17/1  routed      up       up             N/A         N/A
 Ethernet65               78      10G   9100    N/A  Ethernet17/2  routed      up       up             N/A         N/A
 Ethernet66               79      10G   9100    N/A  Ethernet17/3  routed    down       up             N/A         N/A
 Ethernet67               80      10G   9100    N/A  Ethernet17/4  routed    down       up             N/A         N/A
 Ethernet68      73,74,75,76      40G   9100    N/A  Ethernet18/1  routed    down       up             N/A         N/A
 Ethernet72      93,94,95,96      40G   9100    N/A  Ethernet19/1  routed    down       up             N/A         N/A
 Ethernet76      89,90,91,92      40G   9100    N/A  Ethernet20/1  routed    down       up             N/A         N/A
 Ethernet80  101,102,103,104      40G   9100    N/A  Ethernet21/1  routed    down       up             N/A         N/A
 Ethernet84     97,98,99,100      40G   9100    N/A  Ethernet22/1  routed    down       up             N/A         N/A
 Ethernet88  109,110,111,112      40G   9100    N/A  Ethernet23/1  routed    down       up             N/A         N/A
 Ethernet92  105,106,107,108      40G   9100    N/A  Ethernet24/1  routed    down       up             N/A         N/A
 Ethernet96      61,62,63,64      40G   9100    N/A    Ethernet25  routed    down       up             N/A         N/A
Ethernet100      57,58,59,60      40G   9100    N/A    Ethernet26  routed    down       up             N/A         N/A
Ethernet104      81,82,83,84      40G   9100    N/A    Ethernet27  routed    down       up             N/A         N/A
Ethernet108      85,86,87,88      40G   9100    N/A    Ethernet28  routed    down       up             N/A         N/A
Ethernet112  117,118,119,120      40G   9100    N/A    Ethernet29  routed    down       up             N/A         N/A
Ethernet116  113,114,115,116      40G   9100    N/A    Ethernet30  routed    down       up             N/A         N/A
Ethernet120          5,6,7,8      40G   9100    N/A    Ethernet31  routed    down       up             N/A         N/A
Ethernet124          1,2,3,4      40G   9100    N/A    Ethernet32  routed    down       up             N/A         N/A
admin@localhost:~$ sfpshow eeprom
[..]

Ethernet12: SFP EEPROM Not detected

Ethernet16: SFP EEPROM detected
        Application Advertisement: N/A
        Connector: No separable connector
        Encoding: 64B66B
        Extended Identifier: Power Class 1(1.5W max)
        Extended RateSelect Compliance: QSFP+ Rate Select Version 1
        Identifier: QSFP+ or later
        Length Cable Assembly(m): 1
        Nominal Bit Rate(100Mbs): 103
        Specification compliance:
                10/40G Ethernet Compliance Code: 40GBASE-CR4
                Fibre Channel Speed: 1200 Mbytes/Sec
                Fibre Channel link length/Transmitter Technology: Electrical inter-enclosure (EL)
                Fibre Channel transmission media: Twin Axial Pair (TW)
        Vendor Date Code(YYYY-MM-DD Lot): 2016-12-29
        Vendor Name: Arista Networks
        Vendor OUI: 00-1c-73
        Vendor PN: QSFP-4SFP10G-DAC
        Vendor Rev: 01
[... 182 lines cut ...]
      xe82( 83)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe83( 84)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe84( 85)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe85( 86)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe86( 87)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe87( 88)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe88( 89)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe89( 90)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe90( 91)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe91( 92)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe92( 93)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe93( 94)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe94( 95)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe95( 96)  !ena   10G  FD None  No   Disable  TX RX   None   FA  XGMII  9412
      xe96( 97)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe97( 98)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe98( 99)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
      xe99(100)  down   40G  FD   SW  No   Forward          None   FA  XGMII  9122
     xe100(101)  down   40G  FD   SW  No   Forward          None   FA    SR4  9122
     xe101(102)  down   40G  FD   SW  No   Forward          None   FA    SR4  9122
     xe102(103)  down   40G  FD   SW  No   Forward          None   FA    SR4  9122
     xe103(104)  down   40G  FD   SW  No   Forward          None   FA    SR4  9122
```

What are your thoughts? Am I correct in my analysis that the SFP cage / I2C location is derived from the ASIC port number?

Staphylo, 2021-04-15:
Hi @bluecmd,

Thanks for reaching out regarding your issue as well as putting time and effort in enabling DPB on one of our products.

Just so you know, we are in the process of enabling DPB on our platforms but haven't looked at the 7050QX-32 yet.
We currently don't have a defined release timeline for DPB but will be happy to help you get past the hurdles you are facing.

### Transceivers

Our implementation for both `sonic_platform.sfp.SfpBase` and `sfputil.py` expects the physical front panel interface index (as numbered on the switch) to get the right Sfp object.
Looking at the `platform.json` in your repo you had the index set properly initially but then changed them with this commit
https://github.com/bluecmd/sonic-buildimage/commit/706089fe5b466097507d78c8644b9d52d341a93e

The index field in `port_config.ini` and `platform.json` refers to the physical front panel port index, it should be fine if different interfaces share the same index.
I would suggest that you change this back to something like you had before (e.g `"20,20,20,20"` for `Ethernet78`)
Let me know if that solves your issue.

In the meantime I'll build an image using your changes and try it for myself.

### Front panel leds

Not sure if you already noticed yet but it's unlikely that front panel leds will be working properly with DPB.
Arista devices control these outside of the ASIC and therefore some dedicated logic is required. It's in our todolist.
This logic is implemented through the `led_control.py` plugin used by the `ledd` daemon in `pmon`.
However the interface of this plugin doesn't provide enough information to set leds, you just get the SONiC interface name (e.g `Ethernet20`)
We have some logic that parses the `port_config.ini` to gets the alias (EOS interface name) from which we can then extract the port index and lanes in use. From there we can apply our led color policy.
Because DPB gets rid of the `port_config.ini` in favor of `platform.json` and `hwsku.json` some changes are required.
We are planning on improving the `sonic_platform` API by adding a better led interface there but no timeline for now.

Let me know if you have further questions.

Thanks,

bluecmd, 2021-04-15:
Hello @Staphylo 

I am super-impressed of your answer speed and kind reception, really nice to see!

You are 100% correct - I reverted the changes to the `index` property and things seem to be working w.r.t. SFP cage mapping.
The switch is remote, so you are also correct in that I would not have noticed any LED misbehavior - but it is also nothing major for our use case right now. However happy to hear that it is being fixed of course!

With this my changes now seem to be working and I will be opening a PR towards the mainline SONiC repository with my cleaned up changes.

Thanks again!

closed 2021-04-15 by bluecmd