aristanetworks/sonic #33: show interfaces transceiver eeprom failing after sonic-buildimage commit 27a6641

SavchukRomanLv, 2021-12-15:
We do see failure in Arista devices when run [test_snmp_phy_entity](https://github.com/Azure/sonic-mgmt/blob/8032a7cee566c7b53c858c62a375f3fdbe2bf40c/tests/snmp/test_snmp_phy_entity.py#L487)
When dive deeper we've found that after sonic-buildimage commit [27a664](https://github.com/Azure/sonic-buildimage/commit/27a6641fb8f8d52abea61af69f3cc509dff46b7b) 'hardware_rev' disappeared from STATE_DB, and trying to execute command `show interfaces transceiver eeprom` returns error 

`Traceback (most recent call last):
  File "/usr/local/bin/sfpshow", line 508, in <module>
    cli()
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/bin/sfpshow", line 485, in eeprom
    sfp.get_eeprom()
  File "/usr/local/lib/python3.9/dist-packages/utilities_common/multi_asic.py", line 148, in wrapped_run_on_all_asics
    func(self,  *args, **kwargs)
  File "/usr/local/bin/sfpshow", line 423, in get_eeprom
    self.output += self.convert_interface_sfp_info_to_cli_output_string(
  File "/usr/local/bin/sfpshow", line 395, in convert_interface_sfp_info_to_cli_output_string
    sfp_info_output = self.convert_sfp_info_to_output_string(sfp_info_dict)
  File "/usr/local/bin/sfpshow", line 287, in convert_sfp_info_to_output_string
    output += '{}{}: {}\n'.format(indent, QSFP_DATA_MAP[key], sfp_info_dict[key])
KeyError: 'hardware_rev'`

Platform info: 
`Platform: x86_64-arista_7170_64c
HwSKU: Arista-7170-64C
ASIC: barefoot`

Staphylo, 2021-12-16:
I'm able to replicate the issue on the latest master image. I'll update this issue once a fix is available.
In the meantime you can use the following command to dirty patch a running image.
`sed -i 's/sfp_info_dict\[key\]/sfp_info_dict.get(key)/' /usr/local/bin/sfpshow /usr/local/lib/python3*/dist-packages/sfputil/main.py`

Staphylo, 2021-12-17:
For more insight, the issue is related to our recent transition to `sonic_xcvr` which is a overhaul of the xcvr management that we have been working on for quite a while.
It appears that the `hardware_rev` key was renamed `vendor_rev` which makes the old and new API incompatible on this property.
Someone from our team is working toward resolving this issue.

andywongarista, 2021-12-17:
The root cause of this issue was addressed in sonic-utilities: https://github.com/Azure/sonic-utilities/pull/1950
However at sonic-buildimage commit 27a664 the sonic-utilities submodule does not have this fix yet. https://github.com/Azure/sonic-utilities/blob/a0bff26a00d63289c0cbd8ff17f5acf930588553/scripts/sfpshow#L41
This error should be resolved once the submodule is updated.

andywongarista, 2022-01-04:
sonic-utilities submodule has been updated as of sonic-buildimage commit b96533 to include the fix so this issue should now be resolved.

closed 2022-01-04 by andywongarista