aristanetworks/sonic #35: DCS-7050QX-32S question

dinosn, 2022-01-07:
Hello, 

  I'm having one problem and I don't seem at the moment to be able to find a solution on Sonic's documentation.  After the installation on my DCS-7050QX-32S, I run arista setup and rebooted the system.  I still have no ports visible on the environment, only the following is showing.  

```
root@sonic:~# ip address list
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/16 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether c0:d6:82:5b:93:c6 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::c2d6:82ff:fe5b:93c6/64 scope link
       valid_lft forever preferred_lft forever
3: bcm0: <BROADCAST,MULTICAST> mtu 9100 qdisc noop state DOWN group default qlen 1000
    link/ether 02:10:18:7b:33:ed brd ff:ff:ff:ff:ff:ff
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 02:42:e9:a9:44:5c brd ff:ff:ff:ff:ff:ff
    inet 240.127.1.1/24 brd 240.127.1.255 scope global docker0
       valid_lft forever preferred_lft forever
    inet6 fd00::1/80 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::1/64 scope link
       valid_lft forever preferred_lft forever
5: Bridge: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 9100 qdisc noqueue state DOWN group default qlen 1000
    link/ether c0:d6:82:5b:93:c6 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3444:d2ff:fea8:641f/64 scope link
       valid_lft forever preferred_lft forever
6: Loopback0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 3a:4c:d7:ea:73:a5 brd ff:ff:ff:ff:ff:ff
    inet 10.1.0.1/32 scope global Loopback0
       valid_lft forever preferred_lft forever
    inet6 fe80::384c:d7ff:feea:73a5/64 scope link
       valid_lft forever preferred_lft forever
7: dummy: <BROADCAST,NOARP> mtu 1500 qdisc noop master Bridge state DOWN group default qlen 1000
    link/ether 52:d8:8a:b6:ab:bb brd ff:ff:ff:ff:ff:ff
```

Any instructions that can be provided to resolve the issue will be appreciated.

Staphylo, 2022-01-11:
Hi @dinosn 

I tried out the latest `master` image and looking through `/var/log/syslog` it seems that `config-setup` is failing to generate a default configuration for the device.
It seems that the `hwsku.json` for the default_sku `Arista-7050-QX-32S` is running into some issues.
Until we come up with a fix you can use a different `HwSku` or edit the default `HwSku`.

You can re-generate a default configuration in the following fashion:
```
sonic-cfggen -H -k Arista-7050QX-32S --preset t1 > /etc/sonic/config_db.json
sync
arista reboot
```

dinosn, 2022-01-11:
Hello, thank you for your update it's really welcome.  I will proceed as per instructions.

closed 2022-01-20 by Staphylo