aristanetworks/sonic #38: logrotate -f failing due to duplicating arista configuration in logrotate.d

OleksandrKozodoi, 2022-01-10:
There is failure in Arista devices when run [test_pfcwd_timer_accuracy](https://github.com/Azure/sonic-mgmt/blob/master/tests/pfcwd/test_pfcwd_timer_accuracy.py).
Changes from [e43c797](https://github.com/aristanetworks/sonic/commit/e43c7977f7473be67a7e8be7846c9dd71357feae#diff-af885944defd2cb4f39f06148b8431de9425d2759d79b75ca78cc5d831d9fea2) commit caused conflict with changes from [a685fe1](https://github.com/Azure/sonic-buildimage/commit/a685fe17650e5f8857902865943f9034d03dcf92) commit of sonic-buildimage repo. This affects the correct work of `logrotate` utility on Arista devices and trying to execute command `logrotate -f /etc/logrotate.conf` returns error
```
$ sudo logrotate -f /etc/logrotate.conf
error: rsyslog:16 duplicate log entry for /var/log/arista.log
error: found error in file rsyslog, skipping
```
Platform info:
```
Platform: x86_64-arista_7170_64c
HwSKU: Arista-7170-64C
ASIC: barefoot
```

Staphylo, 2022-01-10:
Thanks for bringing this issue to our attention.
Because our driver repo now implements its own logrotate configuration we will favor it and revert the change made by https://github.com/Azure/sonic-buildimage/commit/a685fe17650e5f8857902865943f9034d03dcf92

OleksandrKozodoi, 2022-01-10:
@Staphylo It makes sense. Thank you for responding so quickly!

Staphylo, 2022-03-15:
Closing this issue has the fix made its way to sonic-buildimage.