aristanetworks/sonic #39: [chassis] Pcied process exiting on supervisor and linecard

arlakshm, 2022-05-06:
Pcied process exits immediately after startup
Logs
```
/var/log/syslog:May  1 02:11:38.863624 str2-7804-sup-1 NOTICE pmon#pcied[28]: Failed to load platform Pcie module. Error : No module named 'sonic_platform.pcie', Fallback to default module
/var/log/syslog:May  1 02:11:47.376953 str2-7804-sup-1 INFO pmon#supervisord 2022-05-01 02:11:47,376 INFO success: pcied entered RUNNING state, process has stayed up for > than 10 seconds (startsecs)
/var/log/syslog:May  1 02:12:38.865355 str2-7804-sup-1 INFO pmon#/supervisord: pcied Failed to load platform Pcie module. Error : No module named 'sonic_platform.pcie', Fallback to default module
/var/log/syslog:May  1 02:12:38.865355 str2-7804-sup-1 INFO pmon#/supervisord: pcied Error: [Errno 2] No such file or directory: '/usr/share/sonic/platform/pcie.yaml'
/var/log/syslog:May  1 02:12:38.865355 str2-7804-sup-1 INFO pmon#/supervisord: pcied Not found config file, please add a config file manually, or generate it by running [pcieutil pcie_generate]
/var/log/syslog:May  1 02:12:38.908149 str2-7804-sup-1 INFO pmon#supervisord 2022-05-01 02:12:38,907 INFO exited: pcied (exit status 0; expected)
```

Staphylo, 2022-05-09:
We do not implement `sonic_platform.Pcie` plugin but this is in the pipeline.
However the `pcie.yaml` configurations required in the meantime are definetly missing.
This is addressed by https://github.com/Azure/sonic-buildimage/pull/10749

arlakshm, 2022-05-09:
Thanks @Staphylo https://github.com/Azure/sonic-buildimage/pull/10749, has the changes for Linecard. Do we need another PR for supervisor?

Staphylo, 2022-05-10:
I believe the supervisor has the necessary configs.
I'll double check and open another PR for the supervisor if something is missing.

Staphylo, 2022-05-10:
It is indeed missing on the supervisor but we'll implement the `sonic_platform.Pcie` plugin for it.
Because a chassis is dynamic in nature, we can't leverage a hardcoded pcie.yaml configuration.

closed 2022-05-10 by Staphylo

Staphylo, 2022-05-25:
Here is the PR that implements the `Pcie` plugin for supervisor and cards https://github.com/Azure/sonic-buildimage/pull/10922