aristanetworks/sonic #43: Get pcied address from chassis_db failing.

arlakshm, 2022-05-23:
This PR https://github.com/Azure/sonic-buildimage/pull/9681 added logic  to update the config_db with asic pcied address present in CHASSIS_DB. This logic is kicked-in only if a file `use_pci_id_chassis`  is present in the platfrom directory of the supervisor or linecard.

This file `use_pci_id_chassis` this file is not present in the 7808 supervisor or multi-asic linecard directory, so the supervisor or multi asic linecard init fails.   Manually adding pcied addresses to asic.conf or config_db helps recover the device.

Staphylo, 2022-05-24:
I'll check this out

Staphylo, 2022-05-24:
I investigated the supervisor issue and we actually had the change internally for it along a few other ones.
I created https://github.com/Azure/sonic-buildimage/pull/10913 which will address the issue.

Regarding setting `use_pci_id_chassis` on the multi asic linecards, I will discuss this with the team later today.

Staphylo, 2022-05-31:
The team confirmed that the `use_pci_id_chassis` file is indeed required for multi-asic scenarios.
The only SKU that needs it is https://github.com/Azure/sonic-buildimage/tree/master/device/arista/x86_64-arista_7800r3a_36d2_lc
I'll make a PR for it.

Staphylo, 2022-05-31:
I opened https://github.com/Azure/sonic-buildimage/pull/10987 which should close this issue.

closed 2022-09-08 by Staphylo