aristanetworks/sonic #44: [chassis] sai switch create fails on Arista-7800R3A-36DM2-C72

arlakshm, 2022-06-03:
The linecard Arista-7800R3A-36DM2-C72 does not come up with the sonic master image. The orchagent aborts because the sai_switch_create fails with the following errors
```
Jun  3 21:30:15.856526 str2-sonic-lc5-1 ERR syncd0#syncd: [06:00.0] SAI_API_SWITCH:platformInit:1465 Invalid instance id 06:00.0 received.
Jun  3 21:30:15.856526 str2-sonic-lc5-1 CRIT syncd0#syncd: [06:00.0] SAI_API_SWITCH:sai_driver_init:643 Error initializing driver, rc = -1.
Jun  3 21:30:15.856605 str2-sonic-lc5-1 CRIT syncd0#syncd: [06:00.0] SAI_API_SWITCH:brcm_sai_create_switch:1174 initializing SDK failed with error Operation failed (0xfffffff5).
Jun  3 21:30:15.856655 str2-sonic-lc5-1 ERR syncd0#syncd: :- sendApiResponse: api SAI_COMMON_API_CREATE failed in syncd mode: SAI_STATUS_FAILURE
```
The asic is detected, lscpi output show the asic pcied address is valid
```
admin@str2-sonic-lc5-1:~$ lspci | grep -i broadcom
06:00.0 Ethernet controller: Broadcom Inc. and subsidiaries Device 8852 (rev 02)
07:00.0 Ethernet controller: Broadcom Inc. and subsidiaries Device 8852 (rev 02)
10:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM57762 Gigabit Ethernet PCIe (rev 01)
admin@str2-sonic-lc5-1:~$
```

sonic version
```
admin@str2-7804-lc5-1:~$ show vers

SONiC Software Version: SONiC.master.104554-b3e33d4f4
Distribution: Debian 11.3
Kernel: 5.10.0-12-2-amd64
Build commit: b3e33d4f4
Build date: Mon May 30 19:49:17 UTC 2022
Built by: AzDevOps@sonic-build-workers-001K5Z
```
please check if this is a platform issue or not.  let me know if you need more info.

arlakshm, 2022-06-03:
[syslog_sairedis_logs.txt](https://github.com/aristanetworks/sonic/files/8835941/syslog_sairedis_logs.txt)

Staphylo, 2022-06-07:
Thanks for bringing this to our attention.
I'll share your issue with the team and report back.

arlakshm, 2022-06-08:
changed the `switch_id`  from 5 to 6, seeing the same crash.
[syslog_2.txt](https://github.com/aristanetworks/sonic/files/8858057/syslog_2.txt)

kenneth-arista, 2022-07-05:
@arlakshm please close if this is no longer an issue.

arlakshm, 2022-07-06:
Issue not seen anymore, closing this issue.

closed 2022-07-06 by arlakshm