aristanetworks/sonic #46: Support for DCS-7050S switches?

frittentheke, 2022-09-08:
With the 7050QX-32 and 7050QX-32S being supported I was wondering if the DCS-7050S series with it's many SFP+ ports was also supported?

Staphylo, 2022-09-08:
Hi @frittentheke

I am here assuming that you refer to `DCS-7050S-52` or `DCS-7050S-64`.
Both of these use a `Trident+` ASIC which was never supported by the broadcom SAI (which started with `Trident2`).
For various reasons, it is unlikely for us to spend cycles on supporting EOL products.

We have a lot of products starting with the prefix `DCS-7050S` spreading over different different generation of ASICs.
Let me know if that is not the SKU you were thinking about.

closed 2022-09-12 by Staphylo

frittentheke, 2022-09-12:
Thanks @Staphylo for the quick response!

> I am here assuming that you refer to `DCS-7050S-52` or `DCS-7050S-64`. Both of these use a `Trident+` ASIC which was never supported by the broadcom SAI (which started with `Trident2`). For various reasons, it is unlikely for us to spend cycles on supporting EOL products.

Those were exactly the models I was thinking about. Simply because they have mostly SFP+ and would be great for a lab environment. I actually though they'd have a `Trident2 `just like e.g. `DCS-7050QX-32S`.

> We have a lot of products starting with the prefix `DCS-7050S` spreading over different different generation of ASICs. Let me know if that is not the SKU you were thinking about.

Is there any overview of SKUs with the used chipset and specs available anywhere? I've found https://github.com/aristanetworks/sonic#supported-platforms but that likely is not covering all particular SKUs and does not indicate the chipset or feature set one can expect.

As you might read from my question I am currently looking into the used market and being able to quickly look up an offered model would be great.

Staphylo, 2022-09-12:
AFAIK the only Arista switches that you'll find on the after markets at this time will be
 - `7050QX-32` `32x40G` with a `Trident2` and `2GB` of storage
 - `7050QX-32S` `32x40G` with a `Trident2` and `4GB` of flash
 - `7060CX-32S` `32x100G` with a `Tomahawk` and `4GB` of flash

These only have QSFP+/QSFP28 ports and would therefore require breakouts cables if you want SFP+ ports.
You'll see that I also listed the flash space which will unfortunately be a deal breaker with running SONiC in a lab environment.
The latest `master` image available weight 2.2GB once extracted, so I'll let you do the math if you simply want to have 2 images installed for an upgrade.
(If you're not afraid of opening the lid, some people on the web have successfully upgraded the storage device)

You can look at this page if you want to see officially supported devices in SONiC https://github.com/sonic-net/SONiC/wiki/Supported-Devices-and-Platforms
I do not know if there is a page stating which feature set is supported by a given ASIC