aristanetworks/sonic #50: [chassiss][pmon] implement cooling algorithm for chassis

arlakshm, 2022-11-03:
Enchance the cooling algorithm  to work for chassis

rlhui, 2023-02-01:
Hi @arlakshm , wondering what's the priority of this one? And background of this? Thanks.

kenneth-arista, 2023-07-06:
Basic cooling algo has been added via https://github.com/aristanetworks/sonic/commit/302c32be14c1dc2bb0bb5a01f8ff30dbfaab8422. More enhancements will be discussed with the community.

Staphylo, 2023-08-24:
@arlakshm there's been some development in this area.
I'm refactoring our cooling algorithm so it can gather sensor information from `CHASSIS_STATE_DB`.
It should be ready in the coming weeks on `master` and `202305`.

Staphylo, 2023-10-24:
Changes related to the cooling algorithm have merged in master and 202305.
Here is the PR for 202305 https://github.com/sonic-net/sonic-buildimage/pull/16702
The only thing that cannot be taken into account is the linecard transceivers temperatures since it's not available to the supervisor. Ideally these should be populated into `CHASSIS_STATE_DB` as well.
Marking this issue as solved.

closed 2023-10-24 by Staphylo