aristanetworks/sonic #52: [chassis][gbsyncd] Continues syslog messages seen on the  Arista-7800R3-48CQM2-C48 linecard

arlakshm, 2022-11-09:
The following logs messages are continuously printed on the  Arista-7800R3-48CQM2-C48 linecard. Looks like we are printing this message every time we are polling stats. 
```
admin@str2-7804-lc6-1:~$ sudo tail -f /var/log/syslog
Nov  9 05:05:07.176718 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 3] [Port query][2] Response 0x0A9E, detail1 0x0593, detail2 0x0114
Nov  9 05:05:07.176750 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:07.176750 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:07.176750 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 3] [Port query][2] Response 0x0A9E, detail1 0x0593, detail2 0x0114
Nov  9 05:05:07.176772 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:07.176772 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:07.176805 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 3] [Port query][3] Response 0x0A9E, detail1 0x0593, detail2 0x0188
Nov  9 05:05:07.176805 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:07.176839 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:07.176839 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 3] [Port
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd  query][3] Response 0x0A9E, detail1 0x0593, detail2 0x0188
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 4] [Port query][0] Response 0x0A9E, detail1 0x0593, detail2 0x000C
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 4] [Port query][0] Response 0x0A9E, detail1 0x0593, detail2 0x000C
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 4] [Port query][1] Response 0x0A9E, detail1 0x0593, detail2 0x0090
Nov  9 05:05:08.387000 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:08.387084 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
Nov  9 05:05:08.387084 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [Slice 4] [Port query][1] Response 0x0A9E, detail1 0x0593, detail2 0x0090
Nov  9 05:05:08.387084 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_mac_statistics]
Nov  9 05:05:08.387084 str2-7804-lc6-1 INFO gbsyncd#supervisord: syncd [OWL][OWL_get_fec_statistics]
```

arlakshm, 2022-11-09:
@Staphylo, FYI

Staphylo, 2022-11-09:
These messages come from the credo SAI @byu343 could you look into this?

arlakshm, 2022-11-26:
Hi @Staphylo , @byu343, any update on this issue. Debugging issues on this linecard is becoming very difficult.
Any way to supress this on the device?

byu343, 2022-11-28:
@arlakshm, the code to fix it is ready.  To get the fix applied, we need to update the build artifact from 0.7.5 to sonic-v0.8.1 in https://github.com/sonic-net/sonic-buildimage/blob/master/platform/components/docker-gbsyncd-credo.mk. I will email 	jimmyzhai (who did the last change) to see if he can update it. If you have the code and can build it, you can try that way before the new build artifact is published.

arlakshm, 2022-12-02:
Thanks @byu343,  I dont have the code. Let me check with jimmyzhai

Staphylo, 2023-01-17:
Should we close this issue?

kenneth-arista, 2023-01-30:
@arlakshm as Samuel asked earlier, should we close this issue?

kenneth-arista, 2023-02-06:
Closing as resolution already discussed

closed 2023-02-06 by kenneth-arista