aristanetworks/sonic #53: [chassis-LC6] [link flap] disabling fanout port does not shut down corresponding port on LC6

wenyiz2021, 2022-12-02:
when fanout's port is specifically disabled, corresponding port on dut is still up:
```
str2-7260cx3-acs-fan-22#show int Ethernet35/1
Ethernet35/1 is administratively down, line protocol is down (disabled)
```
on dut:
```
admin@str2-7804-lc6-1:~$ show int status
      Interface    Lanes    Speed    MTU    FEC         Alias             Vlan    Oper    Admin             Type    Asym PFC
---------------  -------  -------  -----  -----  ------------  ---------------  ------  -------  ---------------  ----------
      Ethernet0      6,7     100G   9100     rs   Ethernet1/1   PortChannel101      up       up  QSFP28 or later         off
      Ethernet4      2,3     100G   9100     rs   Ethernet2/1   PortChannel101      up       up  QSFP28 or later         off
      Ethernet8      4,5     100G   9100     rs   Ethernet3/1   PortChannel103      up       up  QSFP28 or later         off
     Ethernet12      0,1     100G   9100     rs   Ethernet4/1   PortChannel103      up       up  QSFP28 or later         off
     Ethernet16    14,15     100G   9100     rs   Ethernet5/1   PortChannel105      up       up  QSFP28 or later         off
     Ethernet20    10,11     100G   9100     rs   Ethernet6/1   PortChannel105      up       up  QSFP28 or later         off
     Ethernet24    12,13     100G   9100     rs   Ethernet7/1   PortChannel107      up       up  QSFP28 or later         off
     Ethernet28      8,9     100G   9100     rs   Ethernet8/1   PortChannel107      up       up  QSFP28 or later         off
     Ethernet32    22,23     100G   9100     rs   Ethernet9/1   PortChannel109      up       up  QSFP28 or later         off
     Ethernet36    18,19     100G   9100     rs  Ethernet10/1   PortChannel109      up       up  QSFP28 or later         off
     Ethernet40    20,21     100G   9100     rs  Ethernet11/1  PortChannel1011      up       up  QSFP28 or later         off
     Ethernet44    16,17     100G   9100     rs  Ethernet12/1  PortChannel1011      up       up  QSFP28 or later         off
     Ethernet48    30,31     100G   9100     rs  Ethernet13/1           routed      up       up  QSFP28 or later         off
     Ethernet52    26,27     100G   9100     rs  Ethernet14/1  PortChannel1015      up       up  QSFP28 or later         off
     Ethernet56    28,29     100G   9100     rs  Ethernet15/1  PortChannel1015      up       up  QSFP28 or later         off
     Ethernet60    24,25     100G   9100     rs  Ethernet16/1           routed      up       up  QSFP28 or later         off
     Ethernet64    38,39     100G   9100     rs  Ethernet17/1  PortChannel1019      up       up  QSFP28 or later         off
     Ethernet68    34,35     100G   9100     rs  Ethernet18/1  PortChannel1019      up       up  QSFP28 or later         off
```

Steps to reproduce the issue:
1. manually disable any port on fanout switch, by `config t`, then `interface Ethxx`, `shutdown`
2. do show int Ethxx on fanout, see if the port is disabled
3. do show int status on dut, check if any port is down.

wenyiz2021, 2022-12-02:
issue is seen with .10 image:
commit: https://github.com/sonic-net/sonic-buildimage/commit/9d7c23b8539b53b68b8e755c1eef334704dac0e5

arlakshm, 2022-12-02:
This isssue is seen on` x86_64-arista_7800r3_48cqm2_lc` linecard but not seen on` x86_64-arista_7800r3_48cq2_lc` lnecard

arlakshm, 2022-12-02:
@Staphylo  for viz..

kenneth-arista, 2022-12-02:
Thanks for the update. We also tested on x86_64-arista_7800r3_48cq2_lc and didn't see the problem. We will try on the `x86_64-arista_7800r3_48cqm2_lc` card.

wenyiz2021, 2022-12-09:
```
Hi Rita and Arvind,

I would like to proactively bring awareness of a new Brcm CS case we opened today related to the link flap issue on CL2MS, 

CS00012274433: Local Fault and Remote Fault are not polled by linkscan thread

The above limitation leads to the inability to detect peer link down on the ASIC. There may be other side effects as well. We are following up with Broadcom directly to address this. 

Thanks,
Kenneth
```

kenneth-arista, 2023-01-23:
Can we close this issue? The Brcm SAI fix is already included in SAI_7.1.30. The upstream 202205 branch is using SAI_7.1.31.4.

wenyiz2021, 2023-01-23:
> Can we close this issue? The Brcm SAI fix is already included in SAI_7.1.30. The upstream 202205 branch is using SAI_7.1.31.4.

thanks, I didn't know the updates, will verify and close

closed 2023-01-25 by wenyiz2021