aristanetworks/sonic #54: [chassis] Wolverine linecard does not boot-up on latest 202205 image

arlakshm, 2022-12-06:
The wolverine linecard doesnt bootup on the 202205 image. The ` platform-arista-swss.service` fails at bootup.

```
admin@sonic:~$ sudo systemctl status platform-arista-swss.service
● platform-arista-swss.service - Arista early platform initialization
     Loaded: loaded (/lib/systemd/system/platform-arista-swss.service; enabled;>
     Active: failed (Result: exit-code) since Tue 2022-12-06 20:04:04 UTC; 7s a>
    Process: 3241 ExecStart=/usr/bin/arista -l /var/log/arista.log setup --earl>
   Main PID: 3241 (code=exited, status=1/FAILURE)

Dec 06 20:04:09 sonic systemd[1]: Failed to start Arista early platform initial>
Dec 06 20:04:11 sonic systemd[1]: platform-arista-swss.service: Start request r>
Dec 06 20:04:11 sonic systemd[1]: platform-arista-swss.service: Failed with res>
Dec 06 20:04:11 sonic systemd[1]: Failed to start Arista early platform initial>
Dec 06 20:04:12 sonic systemd[1]: platform-arista-swss.service: Start request r>
Dec 06 20:04:12 sonic systemd[1]: platform-arista-swss.service: Failed with res>
Dec 06 20:04:12 sonic systemd[1]: Failed to start Arista early platform initial>
Dec 06 20:04:12 sonic systemd[1]: platform-arista-swss.service: Start request r>
Dec 06 20:04:12 sonic systemd[1]: platform-arista-swss.service: Failed with res>
Dec 06 20:04:12 sonic systemd[1]: Failed to start Arista early platform initial>
lines 1-16/16 (END)
```

arlakshm, 2022-12-06:
@Staphylo FYI

Staphylo, 2022-12-07:
@arlakshm this should be solved by https://github.com/sonic-net/sonic-buildimage/pull/12968 it just needs to merge
A similar fix for master already merged https://github.com/sonic-net/sonic-buildimage/pull/12967
I mistakenly removed a shipping product while cleaning up prototypes.

closed 2022-12-07 by Staphylo