aristanetworks/sonic #55: [chassis] [linecards] chassis midplane info missing in state db on all LCs

wenyiz2021, 2022-12-07:
```
admin@str2-7804-lc7-1:~$  show chassis modules midplane-status
Key * not found in CHASSIS_MIDPLANE_TABLE table
admin@str2-7804-lc7-1:~$ 
admin@str2-7804-lc7-1:~$ redis-cli -n 6
127.0.0.1:6379[6]> keys CHASSIS_MIDPLANE_TABLE*
(empty array)
127.0.0.1:6379[6]> keys *CHASSIS_MIDPLANE_TABLE*
(empty array)
```

wenyiz2021, 2022-12-07:
@Staphylo can you please add  `CHASSIS_MIDPLANE_TABLE` on LCs that contains SUP? Thanks

Staphylo, 2022-12-08:
@wenyiz2021 It can be done fairly easily however the Linecard has no clue what the Supervisor is nor its status.
Depending on how sonic-mgmt behaves it might be unhappy because the Module information for the supervisor is incomplete. It's up to the test to ignore this properly.

I'm able to get to the following result:

```
root@dut-lc6:~# show chassis modules status 
       Name        Description    Physical-Slot    Oper-Status    Admin-Status
-----------  -----------------  ---------------  -------------  --------------
 LINE-CARD3  7800R3AK-36DM2-LC                6         Online              up
SUPERVISOR0                N/A                1         Online              up
```

```
root@dut-lc6:~# show chassis modules midplane-status 
       Name    IP-Address    Reachability
-----------  ------------  --------------
SUPERVISOR0   127.100.1.1            True
```

```
root@dut-lc6:~# sonic-db-cli STATE_DB hgetall 'CHASSIS_MODULE_TABLE|SUPERVISOR0'
{'desc': 'N/A', 'slot': '1', 'oper_status': 'Online', 'num_asics': '0'}
root@dut-lc6:~# sonic-db-cli STATE_DB hgetall 'CHASSIS_MIDPLANE_TABLE|SUPERVISOR0'
{'ip_address': '127.100.1.1', 'access': 'True'}
```

Does that suit your needs?

wenyiz2021, 2022-12-08:
@Staphylo , yes, what you pasted looks good enough. 
do you mean even `Reachability` is `True` doesn't mean SUP is in good state?

Staphylo, 2022-12-08:
`Reachability` is actually tested so it is accurate.
The hardcoded information is `oper_status` and `desc` and `num_asics` for the supervisor.
But it seems like it doesn't matter.
So I'll go ahead and continue with my change. It should make it to the next submodule update we do.
If it's high priority please let me know and I'll try to do one ASAP.

wenyiz2021, 2022-12-09:
thanks @Staphylo, making it in next submodule is good enough.

Staphylo, 2023-01-17:
Should be fixed by
https://github.com/sonic-net/sonic-buildimage/pull/13398
https://github.com/sonic-net/sonic-buildimage/pull/13399

closed 2023-01-17 by Staphylo