aristanetworks/sonic #56: [chassis] [LC5] missing transceivers in database1

wenyiz2021, 2022-12-08:
o/p of `show int status`:
```
admin@str2-7804-lc5-1:~$ show int status
      Interface            Lanes    Speed    MTU    FEC         Alias             Vlan    Oper    Admin             Type    Asym PFC
---------------  ---------------  -------  -----  -----  ------------  ---------------  ------  -------  ---------------  ----------
      Ethernet0      72,73,74,75     100G   9100     rs   Ethernet1/1   PortChannel102      up       up  QSFP28 or later         off
      Ethernet8      80,81,82,83     100G   9100     rs   Ethernet2/1   PortChannel102      up       up  QSFP28 or later         off
     Ethernet16      88,89,90,91     100G   9100     rs   Ethernet3/1   PortChannel104      up       up  QSFP28 or later         off
     Ethernet24      96,97,98,99     100G   9100     rs   Ethernet4/1   PortChannel104      up       up  QSFP28 or later         off
     Ethernet32  104,105,106,107     100G   9100     rs   Ethernet5/1   PortChannel106      up       up  QSFP28 or later         off
     Ethernet40  112,113,114,115     100G   9100     rs   Ethernet6/1   PortChannel106      up       up  QSFP28 or later         off
     Ethernet48  120,121,122,123     100G   9100     rs   Ethernet7/1   PortChannel108      up       up  QSFP28 or later         off
     Ethernet56  128,129,130,131     100G   9100     rs   Ethernet8/1   PortChannel108      up       up  QSFP28 or later         off
     Ethernet64  136,137,138,139     100G   9100     rs   Ethernet9/1  PortChannel1010      up       up  QSFP28 or later         off
     Ethernet72      64,65,66,67     100G   9100     rs  Ethernet10/1  PortChannel1010      up       up  QSFP28 or later         off
     Ethernet80      56,57,58,59     100G   9100     rs  Ethernet11/1  PortChannel1012      up       up  QSFP28 or later         off
     Ethernet88      48,49,50,51     100G   9100     rs  Ethernet12/1  PortChannel1012      up       up  QSFP28 or later         off
     Ethernet96      40,41,42,43     100G   9100     rs  Ethernet13/1           routed      up       up  QSFP28 or later         off
    Ethernet104      32,33,34,35     100G   9100     rs  Ethernet14/1  PortChannel1016      up       up  QSFP28 or later         off
    Ethernet112      24,25,26,27     100G   9100     rs  Ethernet15/1  PortChannel1016      up       up  QSFP28 or later         off
    Ethernet120      16,17,18,19     100G   9100     rs  Ethernet16/1           routed      up       up  QSFP28 or later         off
    Ethernet128        8,9,10,11     100G   9100     rs  Ethernet17/1  PortChannel1020      up       up  QSFP28 or later         off
    Ethernet136          0,1,2,3     100G   9100     rs  Ethernet18/1  PortChannel1020      up       up  QSFP28 or later         off
    Ethernet144      72,73,74,75     100G   9100     rs  Ethernet19/1           routed      up       up              N/A         off
    Ethernet152      80,81,82,83     100G   9100     rs  Ethernet20/1           routed      up       up  QSFP28 or later         off
    Ethernet160      88,89,90,91     100G   9100     rs  Ethernet21/1           routed      up       up  QSFP28 or later         off
    Ethernet168      96,97,98,99     100G   9100     rs  Ethernet22/1           routed      up       up              N/A         off
    Ethernet176  104,105,106,107     100G   9100     rs  Ethernet23/1           routed      up       up              N/A         off
    Ethernet184  112,113,114,115     100G   9100     rs  Ethernet24/1           routed      up       up  QSFP28 or later         off
    Ethernet192  120,121,122,123     100G   9100     rs  Ethernet25/1           routed      up       up              N/A         off
    Ethernet200  128,129,130,131     100G   9100     rs  Ethernet26/1           routed      up       up              N/A         off
    Ethernet208  136,137,138,139     100G   9100     rs  Ethernet27/1           routed      up       up              N/A         off
    Ethernet216      64,65,66,67     100G   9100     rs  Ethernet28/1           routed      up       up              N/A         off
    Ethernet224      56,57,58,59     100G   9100     rs  Ethernet29/1           routed      up       up              N/A         off
    Ethernet232      48,49,50,51     100G   9100     rs  Ethernet30/1           routed      up       up              N/A         off
    Ethernet240      40,41,42,43     100G   9100     rs  Ethernet31/1           routed      up       up  QSFP28 or later         off
    Ethernet248      32,33,34,35     100G   9100     rs  Ethernet32/1           routed      up       up  QSFP28 or later         off
    Ethernet256      24,25,26,27     100G   9100     rs  Ethernet33/1           routed    down     down              N/A         off
    Ethernet264      16,17,18,19     100G   9100     rs  Ethernet34/1           routed    down     down              N/A         off
    Ethernet272        8,9,10,11     100G   9100     rs  Ethernet35/1           routed    down     down              N/A         off
    Ethernet280          0,1,2,3     100G   9100     rs  Ethernet36/1           routed    down     down              N/A         off
 PortChannel102              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
 PortChannel104              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
 PortChannel106              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
 PortChannel108              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1010              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1012              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1016              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1020              N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
```

```
admin@str2-7804-lc5-1:~$ sudo docker exec database0 redis-cli --raw -n 6 keys TRANSCEIVER_INFO\*
TRANSCEIVER_INFO|Ethernet128
TRANSCEIVER_INFO|Ethernet16
TRANSCEIVER_INFO|Ethernet80
TRANSCEIVER_INFO|Ethernet48
TRANSCEIVER_INFO|Ethernet120
TRANSCEIVER_INFO|Ethernet8
TRANSCEIVER_INFO|Ethernet32
TRANSCEIVER_INFO|Ethernet72
TRANSCEIVER_INFO|Ethernet24
TRANSCEIVER_INFO|Ethernet0
TRANSCEIVER_INFO|Ethernet104
TRANSCEIVER_INFO|Ethernet136
TRANSCEIVER_INFO|Ethernet64
TRANSCEIVER_INFO|Ethernet40
TRANSCEIVER_INFO|Ethernet88
TRANSCEIVER_INFO|Ethernet112
TRANSCEIVER_INFO|Ethernet96
TRANSCEIVER_INFO|Ethernet56
admin@str2-7804-lc5-1:~$ 
admin@str2-7804-lc5-1:~$ 
admin@str2-7804-lc5-1:~$ sudo docker exec database1 redis-cli --raw -n 6 keys TRANSCEIVER_INFO\*
TRANSCEIVER_INFO|Ethernet152
TRANSCEIVER_INFO|Ethernet248
TRANSCEIVER_INFO|Ethernet184
TRANSCEIVER_INFO|Ethernet240
TRANSCEIVER_INFO|Ethernet160
```

error:
```
07/12/2022 08:26:56 transceiver_utils.all_transceivers_detec L0051 INFO   | Interfaces not detected: [u'Ethernet168', u'Ethernet232', u'Ethernet208', u'Ethernet160', u'Ethernet176', u'Ethernet248']
```

o/p of `show version`:
```
admin@str2-7804-lc5-1:~$ show ver

SONiC Software Version: SONiC.20220531C.10
Distribution: Debian 11.5
Kernel: 5.10.0-12-2-amd64
Build commit: 1bcdf1fe67
Build date: Thu Nov 10 05:09:15 UTC 2022
Built by: cloudtest@3307be1fc000002

Platform: x86_64-arista_7800r3a_36dm2_lc
HwSKU: Arista-7800R3A-36DM2-C36
ASIC: broadcom
ASIC Count: 2
Serial Number: SGD21190878
Model Number: 7800R3A-36DM2-LC
Hardware Revision: 2a.00
Uptime: 22:03:28 up 31 min,  1 user,  load average: 5.13, 5.23, 4.75
Date: Thu 08 Dec 2022 22:03:28
```

Staphylo, 2022-12-09:
@wenyiz2021 is there a `xcvrd` traceback in the syslog ?
That's usually the reason why some xcvrs would not show up.
Are all the xcvr eeproms showing up when running `sfputil show eeprom` ?

wenyiz2021, 2022-12-09:
@Staphylo 
on lc5:
```
admin@str2-7804-lc5-1:~$ sudo sfputil show eeprom
SFP index 37 out of range (0-35)
Traceback (most recent call last):
  File "/usr/local/bin/sfputil", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/sfputil/main.py", line 657, in eeprom
    presence = platform_chassis.get_sfp(physical_port).get_presence()
AttributeError: 'NoneType' object has no attribute 'get_presence'
```
other linecards are fine

syslog:
```
/var/log/syslog.5:Dec  7 09:01:29.548195 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd Traceback (most recent call last):
/var/log/syslog.5:Dec  7 09:01:29.548250 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/bin/xcvrd", line 8, in <module>
/var/log/syslog.5:Dec  7 09:01:29.548300 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     sys.exit(main())
/var/log/syslog.5:Dec  7 09:01:29.548350 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 2379, in main
/var/log/syslog.5:Dec  7 09:01:29.548740 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     xcvrd.run()
/var/log/syslog.5:Dec  7 09:01:29.548803 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 2284, in run
/var/log/syslog.5:Dec  7 09:01:29.548853 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     port_mapping_data, retry_eeprom_set = self.init()
/var/log/syslog.5:Dec  7 09:01:29.548901 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 2250, in init
/var/log/syslog.5:Dec  7 09:01:29.549568 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     retry_eeprom_set = post_port_sfp_dom_info_to_db(is_warm_start, port_mapping_data, self.xcvr_table_helper, self.stop_event)
/var/log/syslog.5:Dec  7 09:01:29.549636 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 585, in post_port_sfp_dom_info_to_db
/var/log/syslog.5:Dec  7 09:01:29.549687 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     rc = post_port_sfp_info_to_db(logical_port_name, port_mapping, xcvr_table_helper.get_intf_tbl(asic_index), transceiver_dict, stop_event)
/var/log/syslog.5:Dec  7 09:01:29.549736 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 294, in post_port_sfp_info_to_db
/var/log/syslog.5:Dec  7 09:01:29.549785 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     if not _wrapper_get_presence(physical_port):
/var/log/syslog.5:Dec  7 09:01:29.549833 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd   File "/usr/local/lib/python3.9/dist-packages/xcvrd/xcvrd.py", line 126, in _wrapper_get_presence
/var/log/syslog.5:Dec  7 09:01:29.549881 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd     return platform_chassis.get_sfp(physical_port).get_presence()
/var/log/syslog.5:Dec  7 09:01:29.549930 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd AttributeError: 'NoneType' object has no attribute 'get_presence'
/var/log/syslog.5:Dec  7 09:01:29.592991 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-07 09:01:29,592 INFO exited: xcvrd (exit status 1; not expected)
/var/log/syslog.5:Dec  7 09:01:30.597802 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-07 09:01:30,597 INFO gave up: xcvrd entered FATAL state, too many start retries too quickly
/var/log/syslog.5:Dec  7 09:05:36.300184 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-07 09:05:36,297 INFO spawned: 'xcvrd' with pid 27
/var/log/syslog.5:Dec  7 09:05:46.742820 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-07 09:05:46,741 INFO success: xcvrd entered RUNNING state, process has stayed up for > than 10 seconds (startsecs)
/var/log/syslog.5:Dec  7 09:06:04.082777 str2-7804-lc5-1 INFO pmon#supervisord: xcvrd SFP index 36 out of range (0-35)
```

Staphylo, 2022-12-12:
The issue here is not related to `platform.json` it's related to the `platform_chassis.get_sfp(physical_port)` call.
This error happens when `physical_port` is set to an invalid value.

On this product the number should be within `1-36` so physical port (which is the index field in PORT) is likely invalid
Did you craft the `config_db.json` for this linecard manually?
Could you share the content of the PORT table ?

wenyiz2021, 2022-12-12:
@Staphylo thanks for clarifying.
looks like the recycle and inband ports in asic0 stole 2 indices from asic1. 
Arvind mentioned you had a fix in port_config.ini file, please let us know if PR is ready, thanks!

wenyiz2021, 2022-12-13:
@Staphylo just FYI, not only port_config.ini need a fix, minigraph also need a fix -- inband and recycle ports took the space for 2 external ports:
```
(Pdb) interfaces.keys()
[u'Ethernet152', u'Ethernet200', u'Ethernet168', u'Ethernet192', u'Ethernet184', u'Ethernet232', u'Ethernet208', u'Ethernet160', u'Ethernet224', u'Ethernet176', u'Ethernet144', u'Ethernet248', u'Ethernet216', u'Ethernet240']
(Pdb) mg_ports.keys()
[u'Ethernet-Rec1', u'Ethernet152', u'Ethernet200', u'Ethernet168', u'Ethernet192', u'Ethernet184', u'Ethernet232', u'Ethernet208', u'Ethernet160', u'Ethernet-IB1', u'Ethernet176', u'Ethernet224', u'Ethernet144', u'Ethernet216']
(Pdb) len(mg_ports)
14
(Pdb) len(interfaces)
14
(Pdb) intf
u'Ethernet248'
(Pdb) intf in mg_ports
False
```

closed 2022-12-19 by Staphylo