aristanetworks/sonic #57: [pmon] [x86_64-arista_7800r3a_36dm2_lc] ledd exited too quickly

wenyiz2021, 2022-12-13:
seems due to `/usr/share/sonic/hwsku/port_config.ini` is missing
```
Dec 12 23:52:21.475674 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-12 23:52:21,366 INFO spawned: 'ledd' with pid 25
Dec 12 23:52:21.757499 str2-7804-lc5-1 ERR pmon#ledd[25]: Failed to load ledutil: [Errno 2] No such file or directory: '/usr/share/sonic/hwsku/port_config.ini'
Dec 12 23:52:21.761507 str2-7804-lc5-1 INFO pmon#supervisord: ledd Failed to load ledutil: [Errno 2] No such file or directory: '/usr/share/sonic/hwsku/port_config.ini'
Dec 12 23:52:21.796960 str2-7804-lc5-1 INFO pmon#supervisord 2022-12-12 23:52:21,795 INFO exited: ledd (exit status 1; not expected)

root@str2-7804-lc5-1:/# ls /usr/share/sonic/hwsku/port_config.ini                                      
ls: cannot access '/usr/share/sonic/hwsku/port_config.ini': No such file or directory
root@str2-7804-lc5-1:/# 
```

```
root@str2-7804-lc5-1:/# supervisorctl status  ledd 
ledd                             FATAL     Exited too quickly (process log may have details)

root@str2-7804-lc7-1:/# supervisorctl status
chassis_db_init                  EXITED    Dec 12 04:00 AM
chassisd                         RUNNING   pid 22, uptime 15:58:42
dependent-startup                EXITED    Dec 12 04:00 AM
ledd                             RUNNING   pid 24, uptime 15:58:42
lm-sensors                       EXITED    Dec 12 04:00 AM
psud                             RUNNING   pid 27, uptime 15:58:42
rsyslogd                         RUNNING   pid 17, uptime 15:58:44
supervisor-proc-exit-listener    RUNNING   pid 16, uptime 15:58:45
syseepromd                       RUNNING   pid 28, uptime 15:58:42
thermalctld                      RUNNING   pid 30, uptime 15:58:42
xcvrd                            RUNNING   pid 26, uptime 15:58:42

 

admin@str2-7804-lc6-1:~$ docker exec -it pmon bash
root@str2-7804-lc6-1:/# supervisorctl status
chassis_db_init                  EXITED    Dec 12 04:00 AM
chassisd                         RUNNING   pid 22, uptime 15:59:02
dependent-startup                EXITED    Dec 12 04:00 AM
ledd                             RUNNING   pid 24, uptime 15:59:02
lm-sensors                       EXITED    Dec 12 04:00 AM
psud                             RUNNING   pid 27, uptime 15:59:02
rsyslogd                         RUNNING   pid 17, uptime 15:59:04
supervisor-proc-exit-listener    RUNNING   pid 16, uptime 15:59:05
syseepromd                       RUNNING   pid 29, uptime 15:59:02
thermalctld                      RUNNING   pid 30, uptime 15:59:02
xcvrd                            RUNNING   pid 26, uptime 15:59:02

```

only seen on `x86_64-arista_7800r3a_36dm2_lc`, other linecards are fine

wenyiz2021, 2023-01-07:
@Staphylo can you please check if port_config.ini can be added? thanks

patrickmacarthur, 2023-01-09:
I believe this port_config.ini was added in https://github.com/sonic-net/sonic-buildimage/pull/12951.

closed 2023-01-17 by Staphylo