aristanetworks/sonic #60: [pmon] fail to kill/terminate chassisd process on x86_64-arista_7800_sup

wenyiz2021, 2022-12-20:
failed on sup, other skus(linecards) are fine
```
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 2485, uptime 0:00:14
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon bash -c 'kill -15 2485'
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 2485, uptime 0:00:30
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 2485, uptime 0:00:33
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 2485, uptime 0:00:37
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 2485, uptime 0:07:16

admin@str2-7804-sup-1:~$ show ver

SONiC Software Version: SONiC.C.20220531.13
Distribution: Debian 11.6
Kernel: 5.10.0-18-2-amd64
Build commit: f3164f26cf
Build date: Tue Dec 20 00:50:35 UTC 2022
Built by: cloudtest@24ac8139c000000

Platform: x86_64-arista_7800_sup
HwSKU: Arista-7808R3A-FM
ASIC: broadcom
ASIC Count: 12
Serial Number: SSN20290100
Model Number: DCS-7800-SUP1A
Hardware Revision: 01.0a
Uptime: 20:14:59 up  2:12,  1 user,  load average: 2.54, 2.37, 2.71
Date: Tue 20 Dec 2022 20:14:59

```

wenyiz2021, 2022-12-29:
@Staphylo

Staphylo, 2023-01-04:
@wenyiz2021 we'll look into this (cc @patrickmacarthur)

patrickmacarthur, 2023-01-30:
This should be fixed by https://github.com/sonic-net/sonic-platform-daemons/pull/328

kenneth-arista, 2023-02-06:
Closing as resolution already discussed.

closed 2023-02-06 by kenneth-arista