aristanetworks/sonic #61: [chassis] [all skus] thermal info mising in state db

wenyiz2021, 2022-12-27:
```
-> max_threshold = float(thermal_dict[therm_sensor]['value']['high_threshold'])
(Pdb) therm_sensor
u'TEMPERATURE_INFO|QSFP28 31 temp sensor'
(Pdb) thermal_dict[therm_sensor]
{u'type': u'hash', u'value': {u'low_threshold': u'N/A', u'temperature': u'0.0', u'critical_low_threshold': u'N/A', u'warning_status': u'False', u'timestamp': u'20221227 19:58:39', u'is_replaceable': u'False', u'minimum_temperature': u'0.0', u'critical_high_threshold': u'N/A', u'maximum_temperature': u'0.0', u'high_threshold': u'N/A'}, u'expireat': 1672171151.6979766, u'ttl': -0.001}
```

expected:
```
(Pdb) thermal_dict[therm_sensor]
{u'type': u'hash', u'value': {u'low_threshold': u'10.0', u'temperature': u'40.0', u'critical_low_threshold': u'1.0', u'warning_status': u'False', u'timestamp': u'20221227 20:24:27', u'is_replaceable': u'False', u'minimum_temperature': u'40.0', u'critical_high_threshold': u'100.0', u'maximum_temperature': u'40.0', u'high_threshold': u'100.0'}, u'expireat': 1672172671.5776138, u'ttl': -0.001}
```

wenyiz2021, 2022-12-27:
@Staphylo for viz

Staphylo, 2023-01-04:
@wenyiz2021 we decided to report xcvr temperatures a while ago but it seems to have triggered a lot of grief over time.
I will make a change to disable this behavior and disable xcvr temperature reporting beyond the `get_temperature` method of SfpBase.
Our approach was to attach a `Thermal` object to the `SfpOptoeBase` one. But because xcvr models and vendors do not have the same breadth of support for thermal monitoring I'll disable this.

wenyiz2021, 2023-01-04:
> @wenyiz2021 we decided to report xcvr temperatures a while ago but it seems to have triggered a lot of grief over time. I will make a change to disable this behavior and disable xcvr temperature [...]

thanks @Staphylo, just wonder how does disabling xcvr temperatures help to show all temperature inside redis db? with ` redis-dump -d 6 -y -k "TEMP*"`?

Staphylo, 2023-01-25:
I do not believe it helps with that. But it remove the mismatching number of sensors issue. Making it easier to deal with the issue at hand.

wenyiz2021, 2023-01-26:
closing as issue no longer seen in latest build

closed 2023-01-26 by wenyiz2021