aristanetworks/sonic #62: [chassis] [sup] chassis db thermal numbers mismatch with 'show platform temperature' output

wenyiz2021, 2022-12-28:
output of 'show platform temperature' shows 50 temp sensors:
```
admin@str2-7804-sup-1:/usr/local/lib/python3.9/dist-packages/show$ show platform temp
                                 Sensor    Temperature    High TH    Low TH    Crit High TH    Crit Low TH    Warning          Timestamp
---------------------------------------  -------------  ---------  --------  --------------  -------------  ---------  -----------------
                                   Back           32.5       90.0       0.0            95.0           -5.0      False  20221228 19:40:01
                         Board sensor 1         30.675       85.0       0.0            95.0           -5.0      False  20221228 19:40:01
                         Board sensor 2         30.405       85.0       0.0            95.0           -5.0      False  20221228 19:40:01
                                Exhaust         26.203       85.0       0.0            95.0           -5.0      False  20221228 19:40:01
                               Fap0 AVS          61.81      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                              Fap0 Back           36.0       90.0       0.0           100.0           -5.0      False  20221228 19:40:01
                                 Fap0 C         64.878      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                               Fap0 FAB         64.878      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                             Fap0 Front         30.878       80.0       0.0            90.0           -5.0      False  20221228 19:40:01
                               Fap0 NIF          64.81      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                               Fap1 AVS          56.27      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                              Fap1 Back         34.405       90.0       0.0           100.0           -5.0      False  20221228 19:40:01
                                 Fap1 C         58.675      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                               Fap1 FAB         60.013      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                             Fap1 Front         30.338       80.0       0.0            90.0           -5.0      False  20221228 19:40:01
                               Fap1 NIF         58.878      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                                  Front         28.312       75.0       0.0            85.0           -5.0      False  20221228 19:40:01
                                  Inlet         25.013      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
                                    Mid         34.812       90.0       0.0            95.0           -5.0      False  20221228 19:40:01
                                 Outlet         32.405      100.0       0.0           105.0           -5.0      False  20221228 19:40:01
            Power supply 1 inlet sensor           25.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 1 primary hotspot sensor           45.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 1 secondary hotspot sensor           61.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 2 inlet sensor           26.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 2 primary hotspot sensor           46.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 2 secondary hotspot sensor           64.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 3 inlet sensor           26.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 3 primary hotspot sensor           46.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 3 secondary hotspot sensor           62.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 4 inlet sensor           26.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 4 primary hotspot sensor           48.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 4 secondary hotspot sensor           62.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 5 inlet sensor           25.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 5 primary hotspot sensor           45.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 5 secondary hotspot sensor           61.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 6 inlet sensor           27.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 6 primary hotspot sensor           46.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 6 secondary hotspot sensor           59.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 7 inlet sensor           26.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:00
  Power supply 7 primary hotspot sensor           42.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:00
Power supply 7 secondary hotspot sensor           53.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:00
            Power supply 8 inlet sensor           25.0       65.0       0.0            70.0           -5.0      False  20221228 19:40:01
  Power supply 8 primary hotspot sensor           41.0      115.0       0.0           120.0           -5.0      False  20221228 19:40:01
Power supply 8 secondary hotspot sensor           53.0      120.0       0.0           130.0           -5.0      False  20221228 19:40:01
               Ramon 0 Core (secondary)           49.0       85.0       0.0            95.0           -5.0      False  20221228 19:40:01
                            Ramon 0 PCB          38.54       80.0       0.0            90.0           -5.0      False  20221228 19:40:01
               Ramon 1 Core (secondary)         48.405       85.0       0.0            95.0           -5.0      False  20221228 19:40:01
                            Ramon 1 PCB         39.068       80.0       0.0            90.0           -5.0      False  20221228 19:40:01
                                  Right         30.013       80.0       0.0            90.0           -5.0      False  20221228 19:40:01
                      SFP 1 temp sensor            N/A        N/A       N/A             N/A            N/A      False  20221228 19:40:01
```

while output of `redis-dump -H "127.100.1.1" -p 6380 -d 13 -y -k "TEMP*"` from chassis db gives 196:
```
(Pdb) out_dict.keys()
[u'TEMPERATURE_INFO_7|QSFP28 30 temp sensor', u'TEMPERATURE_INFO_7|QSFP 40 temp sensor', u'TEMPERATURE_INFO_5|QSFP28 14 temp sensor', u'TEMPERATURE_INFO_1|Outlet', u'TEMPERATURE_INFO_5|QSFP28 11 temp sensor', u'TEMPERATURE_INFO_3|QSFP 43 temp sensor', u'TEMPERATURE_INFO_1|Power supply 5 secondary hotspot sensor', u'TEMPERATURE_INFO_3|QSFP 44 temp sensor', u'TEMPERATURE_INFO_3|QSFP 33 temp sensor', u'TEMPERATURE_INFO_3|QSFP 13 temp sensor', u'TEMPERATURE_INFO_1|Power supply 8 secondary hotspot sensor', u'TEMPERATURE_INFO_1|Ramon 1 PCB', u'TEMPERATURE_INFO_7|QSFP 37 temp sensor', u'TEMPERATURE_INFO_3|QSFP 7 temp sensor', u'TEMPERATURE_INFO_7|QSFP28 8 temp sensor', u'TEMPERATURE_INFO_1|Board sensor 1', u'TEMPERATURE_INFO_1|Board sensor 2', u'TEMPERATURE_INFO_7|QSFP28 27 temp sensor', u'TEMPERATURE_INFO_7|QSFP 35 temp sensor', u'TEMPERATURE_INFO_3|QSFP 27 temp sensor', u'TEMPERATURE_INFO_1|Power supply 5 primary hotspot sensor', u'TEMPERATURE_INFO_1|Ramon 0 Core (secondary)', u'TEMPERATURE_INFO_1|Power supply 3 secondary hotspot sensor', u'TEMPERATURE_INFO_3|Fap0 core0', u'TEMPERA [... cut ...] ERATURE_INFO_7|QSFP28 11 temp sensor', u'TEMPERATURE_INFO_1|Ramon 0 PCB', u'TEMPERATURE_INFO_1|Fap1 NIF', u'TEMPERATURE_INFO_1|Power supply 2 inlet sensor', u'TEMPERATURE_INFO_5|QSFP28 29 temp sensor', u'TEMPERATURE_INFO_7|Fap0 core0', u'TEMPERATURE_INFO_5|QSFP28 17 temp sensor', u'TEMPERATURE_INFO_7|QSFP28 28 temp sensor', u'TEMPERATURE_INFO_3|QSFP 28 temp sensor']
```

ideally they both should come from platform.json from all linecards, right? please kindly point the difference and next steps.

wenyiz2021, 2022-12-28:
@Staphylo

Staphylo, 2023-01-04:
@wenyiz2021 we'll look into this

wenyiz2021, 2023-01-24:
closing as issue no longer seen

closed 2023-01-24 by wenyiz2021