aristanetworks/sonic #63: [chassis] [all linecards] thermal1 not getting get_presence and get_low_critical_threshold

wenyiz2021, 2022-12-30:
Hi @Staphylo, for 6 thermal entries, thermal1 is missing implementation in platform_api_conn, for get_presence and get_low_critical_threshold.
As thermal0, 2, 3, 4, 5 is fine with the 2 APIs, can you please prioritize to make it consistent with the thermal1? thanks.

```
103                 presence = thermal.get_presence(platform_api_conn, i)
104  
105                 import pdb; pdb.set_trace()
106  ->             if self.expect(presence is not None, "Unable to retrieve thermal {} presence".format(i)):
107                     if self.expect(isinstance(presence, bool), "Thermal {} presence appears incorrect".format(i)):
108                         self.expect(presence is True, "Thermal {} is not present".format(i))
109  
110             self.assert_expectations()
111  
(Pdb) presence
True
(Pdb) i
0
(Pdb) c
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> PDB set_trace (IO-capturing turned off) >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
> /data/sonic-mgmt-int/tests/platform_tests/api/test_thermal.py(105)test_get_presence()
-> import pdb; pdb.set_trace()
(Pdb) i
1
(Pdb) presence
(Pdb) 
```

```
265  
266                 low_critical_threshold = thermal.get_low_critical_threshold(platform_api_conn, i)
267                 import pdb; pdb.set_trace()
268  
269                 # Ensure the thermal low critical threshold temperature is sane
270  ->             if self.expect(low_critical_threshold is not None, "Unable to retrieve Thermal {} low critical threshold".format(i)):
271                     self.expect(isinstance(low_critical_threshold, float), "Thermal {} low threshold appears incorrect".format(i))
(Pdb) low_critical_threshold
-5.0
(Pdb) i
0
(Pdb) c

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> PDB continue (IO-capturing resumed)
> /data/sonic-mgmt-int/tests/platform_tests/api/test_thermal.py(267)test_get_low_critical_threshold()
-> import pdb; pdb.set_trace()
(Pdb) i
1
(Pdb) list
266                 low_critical_threshold = thermal.get_low_critical_threshold(platform_api_conn, i)
267  ->             import pdb; pdb.set_trace()
268  
(Pdb) low_critical_threshold
(Pdb) 
```

error msg:
```
E           Failed: Unable to retrieve Thermal 1 low critical threshold
err_msg    = 'Unable to retrieve Thermal 1 low critical threshold'

E           Failed: Unable to retrieve thermal 1 presence
err_msg    = 'Unable to retrieve thermal 1 presence'
```

the thermal1 is not getting value from the 2 APIs for all 3 linecards:

Platform: x86_64-arista_7800r3a_36dm2_lc
Platform: x86_64-arista_7800r3_48cq2_lc
Platform: x86_64-arista_7800r3_48cqm2_lc

all with latest SONIC image.

Staphylo, 2023-01-04:
@wenyiz2021 we're aware of this issue and have a fix for this internally. I'll update our platform package soon.

wenyiz2021, 2023-01-04:
> @wenyiz2021 we're aware of this issue and have a fix for this internally. I'll update our platform package soon.

thanks

closed 2023-01-17 by Staphylo