aristanetworks/sonic #64: [chassis] x86_64-arista_7800r3_48cqm2_lc some interfaces take ~10min to come up after reboot

wenyiz2021, 2023-01-05:
After a manual reboot, this linecard takes ~10min for some interfaces to come up, which is very abnormal.
Ideally it should be <5min, also other interfaces come up much faster.
Seems issue on those links:

date after device come up:
```
admin@str2-7804-lc6-1:~$ date
Thu 05 Jan 2023 03:54:18 AM UTC
```

some records:
```
admin@str2-7804-lc6-1:~$ show int status
      Interface    Lanes    Speed    MTU    FEC         Alias             Vlan    Oper    Admin             Type    Asym PFC
---------------  -------  -------  -----  -----  ------------  ---------------  ------  -------  ---------------  ----------
      Ethernet0      6,7     100G   9100     rs   Ethernet1/1   PortChannel101      up       up  QSFP28 or later         off
      Ethernet4      2,3     100G   9100     rs   Ethernet2/1   PortChannel101      up       up  QSFP28 or later         off
      Ethernet8      4,5     100G   9100     rs   Ethernet3/1   PortChannel103    down       up  QSFP28 or later         off
     Ethernet12      0,1     100G   9100     rs   Ethernet4/1   PortChannel103      up       up  QSFP28 or later         off
     Ethernet16    14,15     100G   9100     rs   Ethernet5/1   PortChannel105      up       up  QSFP28 or later         off
     Ethernet20    10,11     100G   9100     rs   Ethernet6/1   PortChannel105      up       up  QSFP28 or later         off
     Ethernet24    12,13     100G   9100     rs   Ethernet7/1   PortChannel107      up       up  QSFP28 or later         off
     Ethernet28      8,9     100G   9100     rs   Ethernet8/1   PortChannel107      up       up  QSFP28 or later         off
     Ethernet32    22,23     100G   9100     rs   Ethernet9/1   PortChannel109      up       up  QSFP28 or later         off
     Ethernet36    18,19     100G   9100     rs  Ethernet10/1   PortChannel109      up       up  QSFP28 or later         off
     Ethernet40    20,21     100G   9100     rs  Ethernet11/1  PortChannel1011    down       up  QSFP28 or later         off
     Ethernet44    16,17     100G   9100     rs  Ethernet12/1  PortChannel1011    down       up  QSFP28 or later         off
     Ethernet48    30,31     100G   9100     rs  Ethernet13/1           routed    down       up  QSFP28 or later         off
     Ethernet52    26,27     100G   9100     rs  Ethernet14/1  PortChannel1015    down       up  QSFP28 or later         off
     Ethernet56    28,29     100G   9100     rs  Ethernet15/1  PortChannel1015    down       up  QSFP28 or later         off
     Ethernet60    24,25     100G   9100     rs  Ethernet16/1           routed    down       up  QSFP28 or later         off
     Ethernet64    38,39     100G   9100     rs  Ethernet17/1  PortChannel1019    down       up  QSFP28 or later         off
     Ethernet68    34,35     100G   9100     rs  Ethernet18/1  PortChannel1019    down       up  QSFP28 or later         off
     Ethernet72    36,37     100G   9100     rs  Ethernet19/1           routed    down       up  QSFP28 or later         off
     Ethernet76    32,33     100G   9100     rs  Ethernet20/1           routed    down       up  QSFP28 or later         off
     Ethernet80    46,47     100G   9100     rs  Ethernet21/1           routed    down       up  QSFP28 or later         off
     Ethernet84    42,43     100G   9100     rs  Ethernet22/1           routed    down       up  QSFP28 or later         off
     Ethernet88    44,45     100G   9100     rs  Ethernet23/1           routed    down       up  QSFP28 or later         off
     Ethernet92    40,41     100G   9100     rs  Ethernet24/1           routed    down       up  QSFP28 or later         off
     Ethernet96    94,95     100G   9100     rs  Ethernet25/1           routed    down       up  QSFP28 or later         off
    Ethernet100    90,91     100G   9100     rs  Ethernet26/1           routed      up       up  QSFP28 or later         off
    Ethernet104    92,93     100G   9100     rs  Ethernet27/1           routed      up       up  QSFP28 or later         off
    Ethernet108    88,89     100G   9100     rs  Ethernet28/1           routed      up       up  QSFP28 or later         off
    Ethernet112    86,87     100G   9100     rs  Ethernet29/1           routed      up       up  QSFP28 or later         off
    Ethernet116    82,83     100G   9100     rs  Ethernet30/1           routed      up       up  QSFP28 or later         off
    Ethernet120    84,85     100G   9100     rs  Ethernet31/1           routed      up       up  QSFP28 or later         off
    Ethernet124    80,81     100G   9100     rs  Ethernet32/1           routed      up       up  QSFP28 or later         off
    Ethernet128    78,79     100G   9100     rs  Ethernet33/1           routed    down     down              N/A         off
    Ethernet132    74,75     100G   9100     rs  Ethernet34/1           routed    down     down              N/A         off
    Ethernet136    76,77     100G   9100     rs  Ethernet35/1           routed    down     down              N/A         off
    Ethernet140    72,73     100G   9100     rs  Ethernet36/1           routed    down     down              N/A         off
    Ethernet144    70,71     100G   9100     rs  Ethernet37/1           routed    down     down              N/A         off
    Ethernet148    66,67     100G   9100     rs  Ethernet38/1           routed    down     down              N/A         off
    Ethernet152    68,69     100G   9100     rs  Ethernet39/1           routed    down     down              N/A         off
    Ethernet156    64,65     100G   9100     rs  Ethernet40/1           routed    down     down              N/A         off
    Ethernet160    62,63     100G   9100     rs  Ethernet41/1           routed    down     down              N/A         off
    Ethernet164    58,59     100G   9100     rs  Ethernet42/1           routed    down     down              N/A         off
    Ethernet168    60,61     100G   9100     rs  Ethernet43/1           routed    down     down              N/A         off
    Ethernet172    56,57     100G   9100     rs  Ethernet44/1           routed    down     down              N/A         off
    Ethernet176    54,55     100G   9100     rs  Ethernet45/1           routed    down     down              N/A         off
    Ethernet180    50,51     100G   9100     rs  Ethernet46/1           routed    down     down              N/A         off
    Ethernet184    52,53     100G   9100     rs  Ethernet47/1           routed    down     down              N/A         off
    Ethernet188    48,49     100G   9100     rs  Ethernet48/1           routed    down     down              N/A         off
 PortChannel101      N/A     200G   9100    N/A           N/A           routed    down       up              N/A         N/A
 PortChannel103      N/A     200G   9100    N/A           N/A           routed    down       up              N/A         N/A
 PortChannel105      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
 PortChannel107      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
 PortChannel109      N/A     200G   9100    N/A           N/A           routed    down       up              N/A         N/A
PortChannel1011      N/A     200G   9100    N/A           N/A           routed    down       up              N/A         N/A
PortChannel1015      N/A     200G   9100    N/A           N/A           routed    down       up              N/A         N/A
[... 107 lines cut ...]
    Ethernet136    76,77     100G   9100     rs  Ethernet35/1           routed    down     down              N/A         off
    Ethernet140    72,73     100G   9100     rs  Ethernet36/1           routed    down     down              N/A         off
    Ethernet144    70,71     100G   9100     rs  Ethernet37/1           routed    down     down              N/A         off
    Ethernet148    66,67     100G   9100     rs  Ethernet38/1           routed    down     down              N/A         off
    Ethernet152    68,69     100G   9100     rs  Ethernet39/1           routed    down     down              N/A         off
    Ethernet156    64,65     100G   9100     rs  Ethernet40/1           routed    down     down              N/A         off
    Ethernet160    62,63     100G   9100     rs  Ethernet41/1           routed    down     down              N/A         off
[... cut ...]
 PortChannel109      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1011      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1015      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
PortChannel1019      N/A     200G   9100    N/A           N/A           routed      up       up              N/A         N/A
admin@str2-7804-lc6-1:~$ date
Thu 05 Jan 2023 04:04:37 AM UTC
```

wenyiz2021, 2023-01-05:
@Staphylo @arlarkshm for viz

wenyiz2021, 2023-01-18:
@Staphylo @kenneth-arista This is causing most reboot tests to fail, is there any clue yet? thanks!

wenyiz2021, 2023-01-18:
> @wenyiz2021 I didn't reply to your issue initially but we've been tracking this internally and are currently searching for the root cause. So far we don't know yet why this happens. We will keep you posted.

thanks Samuel, I'm increasing timeout to 10min to get local run pass, but didn't help pipeline much and runtime has increased

byu343, 2023-01-23:
@wenyiz2021 Could you please send us the syslog from time of reboot to the time that all interfaces are up? Thanks.

wenyiz2021, 2023-01-23:
> @wenyiz2021 Could you please send us the syslog from time of reboot to the time that all interfaces are up? Thanks.

@byu343 I don't see the issue now with latest 202205 build, it takes only 2 mins now for all links to come up
thanks

records:
```
device went back time: 
admin@str2-7804-lc6-1:~$ date
Mon 23 Jan 2023 09:59:26 PM UTC
all links up time:
admin@str2-7804-lc6-1:~$ date
Mon 23 Jan 2023 10:01:23 PM UTC
```

closed 2023-01-23 by wenyiz2021