aristanetworks/sonic #65: Arista 7170-64c Sonic Install

jimthedj65, 2023-01-12:
Hi Guys,

I was able to install sonic onto my switch and go in and manage fans and so on. On reboot it fails to find the boot image.

I installed as follows
cd /mnt/flash
`cp /mnt/usb1/sonic-aboot-broadcom.swi .
Aboot:~# boot /mnt/flash/sonic-aboot-broadcom.swi`

When installing it shows the following 
Aboot# boot /mnt/flash/sonic-aboot-broadcom.swi
Error: TPM_BAD_PARAMETER
TPM: failed to deassert physical presence
Error: TPM_BAD_PARAMETER
TPM: failed to lock physical presence command
1197.26: Cleaning flash content /mnt/flash
1200.93: Generating boot-config, machine.conf and cmdline
1201.02: Installing image under /mnt/flash/image-master.201111-ce88a3818
1201.02: Moving swi to a tmpfs
1202.09: Extracting swi content
1203.12: Extracting platform.tar.gz
1203.16: Extracting dockerfs.tar.gz from swi
df: /tmp/tmp.FmM3Hs/sonic-aboot-broadcom.swi: can't find mount point

It still loads the OS ok and I can login etc but on a reboot I get the following

`df: /tmp/tmp.FmM3Hs/sonic-aboot-broadcom.swi: can't find mount point`

When I reboot the switch it shows the output below and exits to aboot. 

Booting flash:image-master.201111-ce88a3818/.sonic-boot.swiSWI_DEFAULT=flash:image-master.201111-ce88a3818/.sonic-boot.swi
flash:image-master.201111-ce88a3818/.sonic-boot.swiSWI_DEFAULT=flash:image-master.201111-ce88a3818/.sonic-boot.swi not found or not a file

Directory structure looks like this

Aboot# ls -lah
drwxrwx---    7 root     88          4.0K Jan 12 18:33 .
drwxrwxr-x    7 root     88          4.0K Jan 12 18:30 ..
-rw-rw----    1 root     88            24 Jan 12 18:14 .imagehash
-rw-rw----    1 root     88          2.7K Jan 12 18:14 .platforms_asic
-rw-rw----    1 root     88          8.4K Jan 12 18:14 .sonic-boot.swi
-rw-rw----    1 root     88           694 Jan 12 18:14 allowlist_paths.conf
drwxrwxrwx    2 root     88          4.0K Jan 12 18:14 boot
drwx--x---   14 root     0           4.0K Jan 12 18:15 docker
-rw-rw----    1 root     88        384.7M Jan 12 18:14 fs.squashfs
-rw-rw----    1 root     88           953 Jan 12 18:15 kernel-cmdline
drwxr-xr-x    3 1001     1001        4.0K Jan 12 18:15 platform
drwxr-xr-x    8 root     0           4.0K Jan 12 18:15 rw
-rw-rw----    1 root     88          8.0K Jan 12 18:14 swi-signature
-rw-rw----    1 root     88            80 Jan 12 18:14 version
drwxr-xr-x    3 root     0           4.0K Jan 12 18:15 work

Video showing the reboot
[https://www.youtube.com/watch?v=5Xt792lRwEk](url)

Any pointers or assistance would be greatly appreciated.

Staphylo, 2023-01-16:
Hi @jimthedj65,

Your issue here is in the content of the `/mnt/flash/boot-config`
You can see `Booting flash:image-master.201111-ce88a3818/.sonic-boot.swiSWI_DEFAULT=flash:image-master.201111-ce88a3818/.sonic-boot.swi`
It should be `Booting flash:image-master.201111-ce88a3818/.sonic-boot.swi` instead.

I'm not quite sure how it got there but my feeling is that your `boot-config` file was created under windows and uses `crlf` instead of `lf` which could confuse our script as it's not something we've tested.

You could easily fix your boot by running the following command
`echo SWI=flash:image-master.201111-ce88a3818/.sonic-boot.swi > /mnt/flash/boot-config`

closed 2023-01-16 by Staphylo

jimthedj65, 2023-01-18:
Hi Staphylo

Thanks for the response, we actually had a corrupted download and we got a copy of 202205 and all is well.

Thanks