aristanetworks/sonic #66: [all skus] reboo not supported inside pmon

wenyiz2021, 2023-01-24:
Hi @Staphylo 
I'm still seeing this module reboot failure:
```
self = <test_module.TestModuleApi object at 0x7f880451e990>, duthosts = [<MultiAsicSonicHost str2-7804-lc5-1>], enum_rand_one_per_hwsku_hostname = 'str2-7804-lc5-1', localhost = <tests.common.devices.local.Localhost object at 0x7f884e178610>
platform_api_conn = <httplib.HTTPConnection instance at 0x7f880439a230>

    def test_reboot(self, duthosts, enum_rand_one_per_hwsku_hostname, localhost, platform_api_conn):
        reboot_type = 'default'
        reboot_timeout = 300
        for mod_idx in range(self.num_modules):
            mod_name = module.get_name(platform_api_conn, mod_idx)
            if mod_name in self.skip_mod_list:
                logger.info("skipping reboot for module {} ".format(mod_name))
            else:
                module_reboot = module.reboot(platform_api_conn, mod_idx, reboot_type)
>               pytest_assert(module_reboot == "True", "module {} reboot failed".format(mod_idx))
E               Failed: module 0 reboot failed
```

can you please confirm what is missing?
from your previous comment:
```
We had this issue of rebooting a module from pmon initially but we went the extra length to make it supported inside pmon.
You should be able to keep the test running.
```

Staphylo, 2023-01-25:
Hi @wenyiz2021 we have 1 more change pending to enable this functionality.
It's in the last phase of internal review/testing so it should be made available soon.

wenyiz2021, 2023-02-15:
hi @Staphylo I'm still see platform reboot not supported in our nightly, would you please let me know if your internal testing passed?

Staphylo, 2023-02-15:
@patrickmacarthur could you keep this issue updated?

patrickmacarthur, 2023-02-15:
We have a patch internally to enable the platform reboot API. It's still under review and I will let you know when it's ready to go upstream.

Staphylo, 2023-03-01:
Patrick's changes should fix this issues https://github.com/sonic-net/sonic-buildimage/pull/14039
PRs for 202205 and 202211 are also out.

wenyiz2021, 2023-03-06:
> Patrick's changes should fix this issues [sonic-net/sonic-buildimage#14039](https://github.com/sonic-net/sonic-buildimage/pull/14039) PRs for 202205 and 202211 are also out.

thanks @Staphylo our testbed was in bad state last weekend, haven't got latest result to verify