aristanetworks/sonic #68: [Wolverine LC] wrong thermal numbers returned from chassis api

wenyiz2021, 2023-01-24:
```
try:
num_thermals = int(chassis.get_num_thermals(platform_api_conn))
except:
pytest.fail("num_thermals is not an integer")
else:
if num_thermals == 0:
pytest.skip("No thermals found on device")
if duthost.facts.get("chassis"):
expected_num_thermals = len(duthost.facts.get("chassis").get('thermals'))
pytest_assert(num_thermals == expected_num_thermals,
"Number of thermals ({}) does not match expected number ({})"
>                         .format(num_thermals, expected_num_thermals))
E           Failed: Number of thermals (2) does not match expected number (6)
```

wenyiz2021, 2023-01-24:
@Staphylo I believe platform.json is correct as you specified for each of the 6 thermals

Staphylo, 2023-01-25:
I believe the correct number is 2 on Wolverine. The platform.json needs to be updated.
@patrickmacarthur

wenyiz2021, 2023-01-26:
@Staphylo @patrickmacarthur can you please remove the 4 thermal entries in platform.json? 
or let me know what need to be removed, I can create one

patrickmacarthur, 2023-01-26:
I have a patch for this that I will submit upstream today.

patrickmacarthur, 2023-01-26:
PR opened: https://github.com/sonic-net/sonic-buildimage/pull/13524

closed 2023-02-15 by wenyiz2021