aristanetworks/sonic #69: [x86_64-arista_7800r3_48cq2_lc] sfp APIs not implemented

wenyiz2021, 2023-01-25:
following APIs not implemented for sku Arista-7800R3-48CQ2-C48, can you please confirm if they are supported?
```
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_model[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_serial[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_transceiver_info[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_transceiver_bulk_status[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_transceiver_threshold_info[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_rx_los[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_tx_fault[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_temperature[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_voltage[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_tx_bias[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_rx_power[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_get_tx_power[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_reset[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_tx_disable[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_tx_disable_channel[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_lpmode[str2-7804-lc7-1]
FAILED platform_tests/api/test_sfp.py::TestSfpApi::test_power_override[str2-7804-lc7-1]
==================== 17 failed, 6 passed in 780.04 seconds =====================
```

wenyiz2021, 2023-01-25:
> These are passing for us. Do you see the failure consistently? Which modules are you using? @patrickmacarthur

I see it fail consistently with only Arista-7800R3-48CQ2-C48, other 2 skus are fine.
I thought you are testing only on wolverine sku?

patrickmacarthur, 2023-01-30:
I tested this on a DUT with 7800R3-48CQM2-C48 linecards and the SFP tests all passed.

wenyiz2021, 2023-02-15:
closing as issue not seen:
```
2023-02-13T02:41:38.2280252Z =========================== short test summary info ============================
2023-02-13T02:41:38.2281460Z SKIPPED [23] /azp/_work/27/s/tests/platform_tests/api/test_sfp.py:40: skipping for supervisor node
2023-02-13T02:41:38.2283118Z =================== 46 passed, 23 skipped in 2145.50 seconds ===================
```

closed 2023-02-15 by wenyiz2021