aristanetworks/sonic #72: chassisdb.conf missing on all skus

wenyiz2021, 2023-01-25:
```
   def test_thermal_global_state_db(duthosts, enum_supervisor_dut_hostname, tbinfo):
        """
         This test case will verify global state db data on supervisor
         Verify data for all sensors from line cards and fabric cards present in global state db
        """
        duthost = duthosts[enum_supervisor_dut_hostname]
        if duthost.facts['modular_chassis'] == "False":
            pytest.skip("Test skipped applicable to modular chassis only")
>       chassis_db_ip = get_chassis_db_ip(duthost)

duthost    = <MultiAsicSonicHost str2-7804-lc5-1>
duthosts   = [<MultiAsicSonicHost str2-7804-lc5-1>]
enum_supervisor_dut_hostname = 'str2-7804-lc5-1'
tbinfo     = {'auto_recover': 'True', 'comment': 'v-saidia', 'conf-name': 'vms26-t2-7800-1', 'duts': ['str2-7804-lc5-1', 'str2-7804-lc6-1', 'str2-7804-lc7-1', 'str2-7804-sup-1'], ...}

platform_tests/test_thermal_state_db.py:86: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
platform_tests/test_thermal_state_db.py:12: in get_chassis_db_ip
    out = duthost.command("cat /etc/sonic/chassisdb.conf")['stdout_lines']
```

```
admin@str2-7804-lc5-1:~$ cat /etc/sonic/chassisdb.conf
cat: /etc/sonic/chassisdb.conf: No such file or directory

admin@str2-7804-lc6-1:~$ cat /etc/sonic/chassisdb.conf
cat: /etc/sonic/chassisdb.conf: No such file or directory
```

Staphylo, 2023-01-26:
@kenneth-arista @ysmanman

ysmanman, 2023-01-30:
Hi @wenyiz2021  Look at code and it seems we should only install chassisdb.conf at /etc/sonic/chassisdb.conf on sup https://github.com/sonic-net/sonic-buildimage/blob/8c2d8ea4afeecb507350f00fdec9b258033edb5f/dockers/docker-orchagent/docker-init.j2#L52 So this looks a test issue to me.

wenyiz2021, 2023-02-01:
thanks @ysmanman for pointing out, closing issue

closed 2023-02-01 by wenyiz2021