aristanetworks/sonic #73: pcie device table missing on all linecards

wenyiz2021, 2023-01-26:
All linecards are missing pcie device table, SUP is fine:
admin@str2-7804-lc7-1:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
admin@str2-7804-lc7-1:~$ 
admin@str2-7804-lc5-1:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
admin@str2-7804-lc5-1:~$ 
admin@str2-7804-lc6-1:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
admin@str2-7804-lc6-1:~$ 

expected:
admin@str2-7804-sup-1:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
PCIE_DEVICES|status

patrickmacarthur, 2023-01-26:
I am not seeing this issue on my system.

```
admin@cmp206-4:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
PCIE_DEVICES|status
```

What linecard models are you seeing this issue on?

wenyiz2021, 2023-01-26:
> I am not seeing this issue on my system.
> 
> ```
> admin@cmp206-4:~$ redis-cli -n 6 keys '*' | grep PCIE_DEVICES
> PCIE_DEVICES|status
> ```
> 
> What linecard models are you seeing this issue on?

I'm now seeing this on these 2 linecards:
x86_64-arista_7800r3_48cq2_lc
x86_64-arista_7800r3_48cqm2_lc

wenyiz2021, 2023-02-06:
closing as issue no longer seen

I'm seeing PCIE_DEVICES on all linecards

closed 2023-02-06 by wenyiz2021