aristanetworks/sonic #75: 7170-64c

jimthedj65, 2023-01-27:
Hi All,

We have SONiC build SONiC Software Version: SONiC.master.203398-5e4a866e3
Distribution: Debian 11.6
Kernel: 5.10.0-18-2-amd64
Build commit: 5e4a866e3
Build date: Sun Jan 15 15:19:36 UTC 2023
Built by: AzDevOps@vmss-soni000BCY

Platform: x86_64-arista_7170_64c
HwSKU: Arista-7170-64C
ASIC: barefoot
ASIC Count: 1
Serial Number: SGD19451927
Model Number: DCS-7170-64C
Hardware Revision: 02.00
Uptime: 13:55:25 up 5 days, 23:42,  1 user,  load average: 1.02, 1.01, 1.00
Date: Fri 27 Jan 2023 13:55:25

We can list our 64 100GBE ports and can see the management ports on eth0 etc. 

How do we configure the 10GBE SFP+ ports I think these are showing up as backplane ports like so

root@sonic:/sys/devices# lspci |grep Ethernet
01:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM57762 Gigabit Ethernet PCIe (rev 20)
02:00.0 Ethernet controller: Broadcom Inc. and subsidiaries NetXtreme BCM57762 Gigabit Ethernet PCIe (rev 20)
04:00.0 Ethernet controller: Intel Corporation Ethernet Connection X552 10 GbE Backplane
04:00.1 Ethernet controller: Intel Corporation Ethernet Connection X552 10 GbE Backplane

any guidance or pointers appreciated.

Staphylo, 2023-01-30:
Hi @jimthedj65 

The "backplane" ports that you see are not related to the SFP ports.
```
01:00.0 (eth1) -> unused
02:00.0 (eth0) -> management interface 1G
04:00.0 (tmp3) & 04:00.1 (tmp4) -> 2 x 10G links between the CPU and the Tofino ASIC
```

On the topic of the SFP+ ports you should be good by simply appending the following content to your `PORT` table in `config_db.json`
```
	"Ethernet256": {
	    "alias": "Ethernet65",
            "fec": "none",
	    "index": "65",
	    "lanes": "256",
	    "mtu": "9100",
            "pfc_asym": "off",
	    "speed": "10000"
	},
	"Ethernet257": {
	    "alias": "Ethernet66",
            "fec": "none",
	    "index": "66",
	    "lanes": "257",
	    "mtu": "9100",
            "pfc_asym": "off",
	    "speed": "10000"
	}
```

Staphylo, 2023-01-31:
As replied to the sonic-dev mailing list, there is a runtime way to add the ports without restarting the dataplane.
```
sonic-db-cli CONFIG_DB hset "PORT|Ethernet256" alias Ethernet65 fec none index 65 lanes 256 mtu 9100 pfc_asym off speed 10000 admin_status up
sonic-db-cli CONFIG_DB hset "PORT|Ethernet257" alias Ethernet66 fec none index 66 lanes 257 mtu 9100 pfc_asym off speed 10000 admin_status up
```

Closing this issue as the initial problem is solved.

closed 2023-01-31 by Staphylo

jimthedj65, 2023-01-31:
Thanks for clarifying