aristanetworks/sonic #82: [wolverine LC] sfputil show error-status cli not supported

wenyiz2021, 2023-02-21:
```
admin@str2-7804-lc5-1:~$ sudo sfputil show error-status
Traceback (most recent call last):
  File "/usr/local/bin/sfputil", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/sfputil/main.py", line 997, in error_status
    namespaces = multi_asic.get_front_end_namespaces()
  File "/usr/local/lib/python3.9/dist-packages/sonic_py_common/multi_asic.py", line 431, in get_front_end_namespaces
    ns_list = get_all_namespaces()
  File "/usr/local/lib/python3.9/dist-packages/sonic_py_common/multi_asic.py", line 221, in get_all_namespaces
    config_db_handle[namespace] =  connect_config_db_for_ns(namespace)
  File "/usr/local/lib/python3.9/dist-packages/sonic_py_common/multi_asic.py", line 47, in connect_config_db_for_ns
    config_db = swsscommon.ConfigDBConnector(namespace=namespace)
  File "/usr/lib/python3/dist-packages/swsscommon/swsscommon.py", line 1908, in __init__
    super(ConfigDBConnector, self).__init__(use_unix_socket_path = use_unix_socket_path, namespace = namespace)
  File "/usr/lib/python3/dist-packages/swsscommon/swsscommon.py", line 1586, in __init__
    for db_name in self.get_db_list():
  File "/usr/lib/python3/dist-packages/swsscommon/swsscommon.py", line 1522, in get_db_list
    return _swsscommon.SonicV2Connector_Native_get_db_list(self)
RuntimeError: :- validateNamespace: Initialize global DB config using API SonicDBConfig::initializeGlobalConfig
```

patrickmacarthur, 2023-02-22:
We are looking into this

patrickmacarthur, 2023-02-23:
What model linecard is this on? I cannot reproduce this issue on our system.

Staphylo, 2023-02-23:
This looks like a multi asic issue so I would guess it's a Wolverine card. I'll let Wenyi confirm.

wenyiz2021, 2023-02-23:
> This looks like a multi asic issue so I would guess it's a Wolverine card. I'll let Wenyi confirm.

@Staphylo @patrickmacarthur this is wolverine LC, I specified in this title

wenyiz2021, 2023-03-06:
@patrickmacarthur are you able to reproduce this on wolverine linecard? please let me know, thanks

wenyiz2021, 2023-04-13:
closing as passed latetest run

[image]