aristanetworks/sonic #83: [wolverine LC] watchdog reboot-cause not supported

wenyiz2021, 2023-02-21:
```
20/02/2023 10:01:42 reboot.check_reboot_cause                L0279 DEBUG  | dut str2-7804-lc5-1 last reboot-cause Unknown
20/02/2023 10:01:42 utilities.wait_until                     L0127 DEBUG  | check_reboot_cause is False, wait 20 seconds and check again
20/02/2023 10:02:02 utilities.wait_until                     L0132 DEBUG  | check_reboot_cause is still False after 120 seconds, exit with False
20/02/2023 10:02:02 __init__.pytest_runtest_call             L0040 ERROR  | Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/_pytest/python.py", line 1464, in runtest
    self.ihook.pytest_pyfunc_call(pyfuncitem=self)
  File "/usr/local/lib/python2.7/dist-packages/pluggy/hooks.py", line 286, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pluggy/manager.py", line 93, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pluggy/manager.py", line 87, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python2.7/dist-packages/pluggy/callers.py", line 208, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python2.7/dist-packages/pluggy/callers.py", line 81, in get_result
    _reraise(*ex)  # noqa
  File "/usr/local/lib/python2.7/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python2.7/dist-packages/_pytest/python.py", line 174, in pytest_pyfunc_call
    testfunction(**testargs)
  File "/azp/_work/58/s/tests/platform_tests/test_reboot.py", line 275, in test_watchdog_reboot
    reboot_and_check(localhost, duthost, conn_graph_facts["device_conn"][duthost.hostname], xcvr_skip_list, REBOOT_TYPE_WATCHDOG)
  File "/azp/_work/58/s/tests/platform_tests/test_reboot.py", line 69, in reboot_and_check
    check_interfaces_and_services(dut, interfaces, xcvr_skip_list, reboot_type)
  File "/azp/_work/58/s/tests/platform_tests/test_reboot.py", line 115, in check_interfaces_and_services
    "got reboot-cause failed after rebooted by %s" % reboot_type
AssertionError: got reboot-cause failed after rebooted by watchdog
```

wenyiz2021, 2023-02-21:
@Staphylo @patrickmacarthur

patrickmacarthur, 2023-02-22:
We are looking into this

wenyiz2021, 2023-02-27:
@arlakshm for viz

wenyiz2021, 2023-03-06:
@patrickmacarthur please let us know if any progress is made, thanks

patrickmacarthur, 2023-03-08:
I am working on a fix for this issue

wenyiz2021, 2023-04-13:
still fail on test_watchdog_reboot on wolverine card

patrickmacarthur, 2023-04-18:
We have a fix internally for this and will open a PR for this soon.

Staphylo, 2023-04-24:
PRs are opened for various releases:
 - https://github.com/sonic-net/sonic-buildimage/pull/14827
 - https://github.com/sonic-net/sonic-buildimage/pull/14828
 - https://github.com/sonic-net/sonic-buildimage/pull/14829