aristanetworks/sonic #84: [chassis] got 69 thermal sensors expected 63

wenyiz2021, 2023-02-27:
test_thermal_global_state_db failing on sup, seems due to 6 more entries in one of the platform.json
error:
```
pytest_assert(actual_num_thermal_sensors == expected_num_thermals,
>                     "got {} thermal sensors expected {}".format(actual_num_thermal_sensors, expected_num_thermals))
E       Failed: got 69 thermal sensors expected 63
actual_num_thermal_sensors = 69
chassis_db_ip = '127.100.1.1'
duthost    = <MultiAsicSonicHost str2-7804-sup-1>
duthosts   = [<MultiAsicSonicHost str2-7804-lc5-1>, <MultiAsicSonicHost str2-7804-lc3-1>, <MultiAsicSonicHost str2-7804-lc7-1>, <MultiAsicSonicHost str2-7804-sup-1>]
enum_supervisor_dut_hostname = 'str2-7804-sup-1'
```

```
admin@str2-7804-sup-1:~$ redis-dump -H 127.100.1.1  -p 6380 -d 13 -y -k "TEMP*" | grep TEMPERATURE_INFO_ | wc
     69     297    3119
admin@str2-7804-sup-1:~$ redis-dump -H 127.100.1.1  -p 6380 -d 13 -y -k "TEMP*" | grep TEMPERATURE_INFO_
  "TEMPERATURE_INFO_1|Back": {
  "TEMPERATURE_INFO_1|Board sensor 1": {
  "TEMPERATURE_INFO_1|Board sensor 2": {
  "TEMPERATURE_INFO_1|Exhaust": {
  "TEMPERATURE_INFO_1|Fap0 AVS": {
  "TEMPERATURE_INFO_1|Fap0 Back": {
  "TEMPERATURE_INFO_1|Fap0 C": {
  "TEMPERATURE_INFO_1|Fap0 FAB": {
  "TEMPERATURE_INFO_1|Fap0 Front": {
  "TEMPERATURE_INFO_1|Fap0 NIF": {
  "TEMPERATURE_INFO_1|Fap1 AVS": {
  "TEMPERATURE_INFO_1|Fap1 Back": {
  "TEMPERATURE_INFO_1|Fap1 C": {
  "TEMPERATURE_INFO_1|Fap1 FAB": {
  "TEMPERATURE_INFO_1|Fap1 Front": {
  "TEMPERATURE_INFO_1|Fap1 NIF": {
  "TEMPERATURE_INFO_1|Front": {
  "TEMPERATURE_INFO_1|Inlet": {
  "TEMPERATURE_INFO_1|Mid": {
  "TEMPERATURE_INFO_1|Outlet": {
  "TEMPERATURE_INFO_1|Power supply 1 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 1 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 1 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 2 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 2 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 2 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 3 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 3 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 3 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 4 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 4 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 4 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 5 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 5 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 5 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 6 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 6 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 6 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 7 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 7 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 7 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 8 inlet sensor": {
  "TEMPERATURE_INFO_1|Power supply 8 primary hotspot sensor": {
  "TEMPERATURE_INFO_1|Power supply 8 secondary hotspot sensor": {
  "TEMPERATURE_INFO_1|Ramon 0 Core (secondary)": {
  "TEMPERATURE_INFO_1|Ramon 0 PCB": {
  "TEMPERATURE_INFO_1|Ramon 1 Core (secondary)": {
  "TEMPERATURE_INFO_1|Ramon 1 PCB": {
  "TEMPERATURE_INFO_1|Right": {
  "TEMPERATURE_INFO_3|Center back": {
  "TEMPERATURE_INFO_3|Cpu SBTSI": {
  "TEMPERATURE_INFO_3|Cpu temp sensor": {
  "TEMPERATURE_INFO_3|Fap0 core0": {
  "TEMPERATURE_INFO_3|Fap0 core1": {
  "TEMPERATURE_INFO_3|PCIE": {
  "TEMPERATURE_INFO_5|Cpu SBTSI": {
  "TEMPERATURE_INFO_5|Cpu temp sensor": {
  "TEMPERATURE_INFO_6|Center back": {
  "TEMPERATURE_INFO_6|Cpu SBTSI": {
  "TEMPERATURE_INFO_6|Cpu temp sensor": {
  "TEMPERATURE_INFO_6|Fap0 core0": {
  "TEMPERATURE_INFO_6|Fap0 core1": {
  "TEMPERATURE_INFO_6|PCIE": {
  "TEMPERATURE_INFO_7|Center back": {
  "TEMPERATURE_INFO_7|Cpu SBTSI": {
  "TEMPERATURE_INFO_7|Cpu temp sensor": {
  "TEMPERATURE_INFO_7|Fap0 core0": {
  "TEMPERATURE_INFO_7|Fap0 core1": {
  "TEMPERATURE_INFO_7|PCIE": {
```

@Staphylo @patrickmacarthur could you please take a look?
I'm not sure where `TEMPERATURE_INFO_1` come from, the TEMPERATURE_INFO_3/5/6/7 seems to come from corresponding platform.json thermal section

patrickmacarthur, 2023-03-16:
We are looking into this

wenyiz2021, 2023-05-11:
platform_tests/test_thermal_state_db.py::test_thermal_state_db[str2-7804-lc5-1] PASSED [ 25%]
platform_tests/test_thermal_state_db.py::test_thermal_state_db[str2-7804-lc3-1] PASSED [ 50%]
platform_tests/test_thermal_state_db.py::test_thermal_state_db[str2-7804-sup-1] PASSED [ 75%]
platform_tests/test_thermal_state_db.py::test_thermal_global_state_db[str2-7804-sup-1] PASSED [100%]

closed 2023-05-11 by wenyiz2021