aristanetworks/sonic #86: [chassis] [cleawater2] [wolverine] linecards - DB data keys is not cleared on chassisd stop

wenyiz2021, 2023-03-09:
sup is fine with expected output:
```
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         RUNNING   pid 22, uptime 2:34:07
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl stop chassisd
chassisd: stopped
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         STOPPED   Mar 09 10:19 PM
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ sonic-db-cli STATE_DB KEYS "CHASSIS_*TABLE|*"

admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ docker exec pmon supervisorctl start chassisd
chassisd: started
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ 
admin@str2-7804-sup-1:~$ sonic-db-cli STATE_DB KEYS "CHASSIS_*TABLE|*"
CHASSIS_MIDPLANE_TABLE|LINE-CARD4
CHASSIS_TABLE|CHASSIS 1
CHASSIS_MODULE_TABLE|FABRIC-CARD5
CHASSIS_MODULE_TABLE|LINE-CARD5
CHASSIS_MODULE_TABLE|SUPERVISOR0
CHASSIS_MODULE_TABLE|FABRIC-CARD2
CHASSIS_MODULE_TABLE|FABRIC-CARD0
CHASSIS_MIDPLANE_TABLE|LINE-CARD0
CHASSIS_MODULE_TABLE|FABRIC-CARD4
CHASSIS_MIDPLANE_TABLE|LINE-CARD2
CHASSIS_MODULE_TABLE|FABRIC-CARD3
CHASSIS_MODULE_TABLE|LINE-CARD0
CHASSIS_MIDPLANE_TABLE|LINE-CARD1
CHASSIS_MIDPLANE_TABLE|LINE-CARD3
CHASSIS_MODULE_TABLE|LINE-CARD7
CHASSIS_MODULE_TABLE|LINE-CARD3
CHASSIS_MIDPLANE_TABLE|LINE-CARD6
CHASSIS_MODULE_TABLE|LINE-CARD6
CHASSIS_MODULE_TABLE|LINE-CARD2
CHASSIS_MIDPLANE_TABLE|LINE-CARD7
CHASSIS_MODULE_TABLE|LINE-CARD1
CHASSIS_MIDPLANE_TABLE|LINE-CARD5
CHASSIS_MODULE_TABLE|FABRIC-CARD1
CHASSIS_MODULE_TABLE|LINE-CARD4
```

error output:
```
admin@str2-7804-lc7-1:~$ docker exec pmon supervisorctl status chassisd
chassisd                         STOPPED   Mar 09 09:54 PM
admin@str2-7804-lc7-1:~$  sonic-db-cli STATE_DB KEYS "CHASSIS_*TABLE|*"
CHASSIS_MODULE_TABLE|SUPERVISOR0
CHASSIS_MIDPLANE_TABLE|SUPERVISOR0
CHASSIS_MODULE_TABLE|LINE-CARD
CHASSIS_TABLE|CHASSIS 1
admin@str2-7804-lc7-1:~$ 
```

this fails `platform_tests/daemon/test_chassisd.py::test_pmon_chassisd_stop_and_start_status`
issue linecards:
1. HwSKU: Arista-7800R3-48CQ2-C48
2. HwSKU: Arista-7800R3A-36DM2-D36

patrickmacarthur, 2023-03-09:
We have seen this issue locally; I am working on a fix.

patrickmacarthur, 2023-03-16:
https://github.com/sonic-net/sonic-platform-daemons/pull/347 has a fix for this issue.

wenyiz2021, 2023-04-13:
with .24 image passed

closed 2023-04-13 by wenyiz2021