aristanetworks/sonic #95: Support for DCS-7124SX

mentax007, 2023-07-20:
Hello, 
I have an Arista DCS-7124SX and wonder if I can use sonic on this platform. I can't find any mentions about this platform in wiki or anywhere in this repo...

Staphylo, 2023-07-20:
Hi @mentax007,

This product is not supported in SONiC and will not receive any official SONiC support.
The main reasons being this are:
 - This product is `EOL`
 - There is no Asic vendor `SAI` available for this line of product (which has been discontinued)
 - Some hardware limitations such as storage space and RAM

mentax007, 2023-07-20:
So, without official support, if I will upgrade ram and add ssd to this platform, is there any way SONiC will be able to run on this hardware?
Because of the EOL I'm looking for newer firmware which could be run on DCS-7124SX. The old version was EOL for a years now and I bet that it is contains whole bunch of bugs which will allow hackers to access the switch.

Staphylo, 2023-07-20:
Even if you take care of the storage/ram limitation you'll still be left off the 2 other main issues:

1) In SONIC the ASIC support comes from the asic vendor.
In the case of this product this is Intel for the FM4000 chip.
However given that this ASIC (as well as the entire product line) has been discontinued it is highly unlikely to happen.

I could stop there since without ASIC support, there is no point in adding platform support.

2) In SONiC the platform support comes the switch vendor.
However as I mentioned previously we will not be spending cycles working on an EOL product.
For perspective this hardware platform was already out of software support when SONiC was first released.