aristanetworks/sonic #98: Arista 7060X6 Sfputil reset does not reset the optics

prgeor, 2024-05-13:
Please validate the behavior of 

1. By default if Xcvrd is not coming up, the module must remain in Low power mode
2. If the module datapath are activated, Sfputil reset must bring the module back to Low power mode.

prgeor, 2024-05-13:
root@str-7060X6-D10-U32:~# show ver

SONiC Software Version: SONiC.internal.92843184-cb03eddfa6
SONiC OS Version: 12
Distribution: Debian 12.5
Kernel: 6.1.0-11-2-amd64
Build commit: cb03eddfa6
Build date: Fri May  3 15:47:29 UTC 2024
Built by: cloudtest@76ca1a58c000000

Platform: x86_64-arista_7060x6_64de
HwSKU: Arista-7060X6-64DE
ASIC: broadcom
ASIC Count: 1
Serial Number: JAS23500161
Model Number: 7001-SUP-A
Hardware Revision: 01.00
Uptime: 23:03:35 up 5 days, 19:39,  3 users,  load average: 0.32, 0.33, 0.42
Date: Mon 13 May 2024 23:03:35

Docker images:
REPOSITORY                 TAG                            IMAGE ID       SIZE
docker-mux                 internal.92843184-cb03eddfa6   2efeb4d62d00   354MB
docker-mux                 latest                         2efeb4d62d00   354MB
docker-macsec              latest                         38004e688e7b   332MB
docker-sonic-telemetry     internal.92843184-cb03eddfa6   fd46776533ea   372MB
docker-sonic-telemetry     latest                         fd46776533ea   372MB
docker-gbsyncd-broncos     internal.92843184-cb03eddfa6   4f14074627c1   354MB
docker-gbsyncd-broncos     latest                         4f14074627c1   354MB
docker-gbsyncd-credo       internal.92843184-cb03eddfa6   0d19d7190048   326MB
docker-gbsyncd-credo       latest                         0d19d7190048   326MB
docker-sonic-gnmi          internal.92843184-cb03eddfa6   433e6ed586ba   371MB
docker-sonic-gnmi          latest                         433e6ed586ba   371MB
docker-orchagent           internal.92843184-cb03eddfa6   48aaf2ceb2ee   342MB
docker-orchagent           latest                         48aaf2ceb2ee   342MB
docker-teamd               internal.92843184-cb03eddfa6   c13b1214597b   330MB
docker-teamd               latest                         c13b1214597b   330MB
docker-dhcp-relay          latest                         f5bdf2cc29da   313MB
docker-snmp                internal.92843184-cb03eddfa6   03c06f8db4d1   341MB
docker-snmp                latest                         03c06f8db4d1   341MB
docker-platform-monitor    internal.92843184-cb03eddfa6   b78d343f34c0   425MB
docker-platform-monitor    latest                         b78d343f34c0   425MB
docker-router-advertiser   internal.92843184-cb03eddfa6   b23695166ec2   302MB
docker-router-advertiser   latest                         b23695166ec2   302MB
docker-lldp                internal.92843184-cb03eddfa6   9c8721e03fa7   348MB
docker-lldp                latest                         9c8721e03fa7   348MB
docker-database            internal.92843184-cb03eddfa6   1d2495e8edc8   310MB
docker-database            latest                         1d2495e8edc8   310MB
docker-eventd              internal.92843184-cb03eddfa6   9a33301ac7d8   302MB
docker-eventd              latest                         9a33301ac7d8   302MB
docker-sonic-restapi       internal.92843184-cb03eddfa6   ee82e9580e6e   303MB
docker-sonic-restapi       latest                         ee82e9580e6e   303MB
docker-fpm-frr             internal.92843184-cb03eddfa6   bb70e8314f45   356MB
docker-fpm-frr             latest                         bb70e8314f45   356MB
docker-syncd-brcm          internal.92843184-cb03eddfa6   349c190c53d5   717MB
docker-syncd-brcm          latest                         349c190c53d5   717MB
docker-acms                internal.92843184-cb03eddfa6   17982b6afd48   350MB
docker-acms                latest                         17982b6afd48   350MB

root@str-7060X6-D10-U32:~#

prgeor, 2024-05-13:
>>> api.get_datapath_state()
{'DP1State': 'DataPathActivated', 'DP2State': 'DataPathActivated', 'DP3State': 'DataPathActivated', 'DP4State': 'DataPathActivated', 'DP5State': 'DataPathActivated', 'DP6State': 'DataPathActivated', 'DP7State': 'DataPathActivated', 'DP8State': 'DataPathActivated'}
>>>

mihirpat1, 2024-08-21:
@prgeor Issued sfputil reset with `20231110.19` image and observed that after issuing sfputil reset

1. Module transitions to low power moder
2. DataPath state transitions to DPDeactivated state.

```
root@str3-7060x6-64pe-3:/home/admin# sfputil read-eeprom -p Ethernet0 -n 0 -o 3 -s 1
        00000003 06                                               |.|
root@str3-7060x6-64pe-3:/home/admin# sfputil read-eeprom -p Ethernet0 -n 11 -o 128 -s 4
        00000080 44 44 44 44                                      |DDDD|
root@str3-7060x6-64pe-3:/home/admin# ### After performing sfputil reset
root@str3-7060x6-64pe-3:/home/admin# sfputil reset Ethernet0
Resetting port Ethernet0 ... OK
root@str3-7060x6-64pe-3:/home/admin# sfputil read-eeprom -p Ethernet0 -n 0 -o 3 -s 1
        00000003 02                                               |.|
root@str3-7060x6-64pe-3:/home/admin# sfputil read-eeprom -p Ethernet0 -n 11 -o 128 -s 4
        00000080 11 11 11 11                                      |....|
root@str3-7060x6-64pe-3:/home/admin#

root@str3-7060x6-64pe-3:/home/admin# show version

SONiC Software Version: SONiC.20231110.19
SONiC OS Version: 11
Distribution: Debian 11.10
Kernel: 5.10.0-23-2-amd64
Build commit: 29d024f4c5
Build date: Wed Aug  7 08:14:09 UTC 2024
Built by: azureuser@cfd1eb31c000000

Platform: x86_64-arista_7060x6_64pe
HwSKU: Arista-7060X6-64PE-256x200G
ASIC: broadcom
ASIC Count: 1
Serial Number: HBG240904VN
Model Number: DCS-7060X6-64PE
Hardware Revision: 02.00
Uptime: 18:09:04 up 30 min,  1 user,  load average: 1.09, 1.34, 1.31
Date: Wed 21 Aug 2024 18:09:04
```

Staphylo, 2024-08-22:
The fix from https://github.com/aristanetworks/sonic/commit/522eb5cca9de7ddd6b96c5a361913855ad5d8e34 will keep the xcvrs in low power mode.

However I am not convinced on the expected behavior for reset.
The requirement that the `reset` method from `SfpBase` has to put the xcvr back into low power mode feels like an overreach and pushing complexity down to every vendor implementation instead of the common code.
The platform API should really only abstract the hardware away where necessary and provide primitives.
This is likely to lead to inconsistency across vendors, which is likely already the case.

I believe it would be much more sensible to have the common code do something like this instead:
```
sfp = chassis.get_sfp(port_id)
sfp.set_low_power_mode(True)
sfp.reset(True)
sfp.set_low_power_mode(False)
```

If going with a common implementation in xcvrd is not desired then we could add a knob to put xcvr back into low power mode during a reset. It is a pretty straightforward change to make.

prgeor, 2024-08-23:
> The fix from [522eb5c](https://github.com/aristanetworks/sonic/commit/522eb5cca9de7ddd6b96c5a361913855ad5d8e34) will keep the xcvrs in low power mode.
> 
> However I am not convinced on the expected behavior for reset. The requirement that the `reset` method from `SfpBase` has to put the xcvr back into low power mode feels like an overreach and pushing complexity down to every vendor implementation instead of the common code. The platform API should really only abstract the hardware away where necessary and provide primitives. This is likely to lead to inconsistency across vendors, which is likely already the case.
> 
> I believe it would be much more sensible to have the common code do something like this instead:
> 
> ```
> sfp = chassis.get_sfp(port_id)
> sfp.set_low_power_mode(True)
> sfp.reset(True)
> sfp.set_low_power_mode(False)
> ```
> 
> If going with a common implementation in xcvrd is not desired then we could add a knob to put xcvr back into low power mode during a reset. It is a pretty straightforward change to make.

@Staphylo the expected behavior has not changed for resetting the optics across all platform/vendors.

When we reset the optics, the expectation is the optics remain in Low power mode unless Xcvrd or other SW entity initialize the optics and bring it out of Low power mode to High power mode. I hope this clarifies why we need to keep the optics in low power mode and under SW control.

prgeor, 2024-08-23:
@Staphylo I must admit, the behavior for your platform previously was to bring the optics out of low power mode to high power mode, but for CMIS optics we don't desire that as optics data path need to be initialized only when there is a valid Tx signal form ASIC to optics. This is ensured by the Xcvrd.

Staphylo, 2024-08-26:
I will make a change to set the hardware low power mode a part of the reset process of a xcvr.

prgeor, 2024-08-26:
@mihirpat1 FYI

prgeor, 2024-08-26:
@Staphylo do you want me to open this issue in https://github.com/aristanetworks/sonic-qual.msft/issues ?

Staphylo, 2024-08-27:
I don't believe this is necessary.
There are 2 issues there related to lpmode and I think they all (including this one) stem from the same problem.
We can continue discussing over there.

prgeor, 2024-09-26:
> @prgeor Issued sfputil reset with `20231110.19` image and observed that after issuing sfputil reset
>
> 1. Module transitions to low power moder
> 2. DataPath state transitions to DPDeactivated state. [...]

closing this as per Mihir's test