aristanetworks/sonic #101: CW2 LC link issue

wenyiz2021, 2024-09-04:
there are 2 links that are down/up.
the following work around worked according to Nathan to disable low power mode:

sudo config interface  tran lpmode Ethernet104 disable

sudo config interface  tran lpmode Ethernet108 disable

but by default the low power mode is enabled.

open this issue to root cause.

kenneth-arista, 2024-09-04:
@Staphylo @arista-nwolfe

kenneth-arista, 2024-09-04:
Related to the lower power mode discussion

kenneth-arista, 2024-11-21:
Related PRs
https://github.com/sonic-net/sonic-buildimage/pull/20886
https://github.com/sonic-net/sonic-platform-daemons/pull/565

wenyiz2021, 2024-12-09:
@rlhui this is still an issue regarding cable type 100GBASE-CWDM4, they need to be set to low power mode to be up now, as WA

wenyiz2021, 2024-12-09:
100GBASE-CR4 old cable type is good

kenneth-arista, 2024-12-18:
Corresponding PRs merged. Closing issue

closed 2024-12-18 by kenneth-arista