aristanetworks/sonic #103: All interface disabled on DCS-7060CX-32S-R

gc-nathanh, 2024-09-27:
We have been investigating running SONiC on some DCS-7060CX-32S switches, which from the list of supported platforms should work.      We've taken a switch that was previously running EOS, and put SONiC on it.   Although the controlplane is up and accessible, none of the switch ports are currently up/usable. 

Taking Ethernet124 as an example:

```
admin@sonic:~$ show int status Ethernet124
  Interface        Lanes    Speed    MTU    FEC         Alias    Vlan    Oper    Admin             Type    Asym PFC
-----------  -----------  -------  -----  -----  ------------  ------  ------  -------  ---------------  ----------
Ethernet124  29,30,31,32      N/A   9100    N/A  Ethernet32/1  routed     N/A       up  QSFP28 or later         N/A
```
which is showing as !ena from bcmcmd ps:

```
ps
                 ena/        speed/ link auto    STP                  lrn  inter   max   cut   loop
           port  link  Lns   duplex scan neg?   state   pause  discrd ops   face frame  thru?  back
       ce0(  1)  !ena   4  100G  FD   SW  No   Forward          None   FA    KR4  9412    No
       ce1(  2)  !ena   4  100G  FD   SW  No   Forward          None   FA    KR4  9412    No
...
...
      ce30(108)  !ena   4  100G  FD   SW  No   Forward          None   FA    KR4  9412    No
      ce31(109)  !ena   4  100G  FD   SW  No   Forward          None   FA    KR4  9412    No
```

We are new to SONiC so it is quite possible that this is administrator error, but we have another switch from a different vendor where the interfaces are all enabled on boot.  Are there additional steps required on Arista platforms post installation?

gc-nathanh, 2024-10-04:
Closing as the issue is elsewhere.   When SONiC was installed on one of the leaf switches, the host facing interfaces, along with the interface running SONIC came up.   The only interfaces that remain down are those connecting to switches still running EOS (which was all of them in the case of my original switch).

What remains to find is what config incompatibility there is preventing the interfaces coming up between switches running SONIC and EOS

closed 2024-10-04 by gc-nathanh

Staphylo, 2024-10-11:
@gc-nathanh did you make sure `fec` is set to `rs` on the SONiC side?