aristanetworks/sonic #104: [7060-X6] Port LEDs are not glowing green when port is UP

prgeor, 2024-10-23:
SONiC version is 202311. 

1. May I know if 7060-X6 has the necessary driver to support port LEDs
2. I will share more details once I get the production switch

prgeor, 2024-10-23:
@Staphylo fyi

prgeor, 2024-10-23:
Links that are UP

```
root@FRA26-0101-0809-01BS0:~# show int status | grep "up       up"
  Ethernet0    17,18     200G   9100     rs   Ethernet1/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
  Ethernet2    19,20     200G   9100     rs   Ethernet1/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
  Ethernet4    21,22     200G   9100     rs   Ethernet1/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
  Ethernet6    23,24     200G   9100     rs   Ethernet1/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
  Ethernet8      1,2     200G   9100     rs   Ethernet2/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet10      3,4     200G   9100     rs   Ethernet2/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet12      5,6     200G   9100     rs   Ethernet2/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet14      7,8     200G   9100     rs   Ethernet2/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet16     9,10     200G   9100     rs   Ethernet3/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet18    11,12     200G   9100     rs   Ethernet3/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet20    13,14     200G   9100     rs   Ethernet3/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet22    15,16     200G   9100     rs   Ethernet3/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet24    25,26     200G   9100     rs   Ethernet4/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet26    27,28     200G   9100     rs   Ethernet4/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet28    29,30     200G   9100     rs   Ethernet4/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet30    31,32     200G   9100     rs   Ethernet4/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet32    57,58     200G   9100     rs   Ethernet5/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet34    59,60     200G   9100     rs   Ethernet5/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet36    61,62     200G   9100     rs   Ethernet5/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet38    63,64     200G   9100     rs   Ethernet5/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet40    41,42     200G   9100     rs   Ethernet6/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet42    43,44     200G   9100     rs   Ethernet6/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet44    45,46     200G   9100     rs   Ethernet6/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet46    47,48     200G   9100     rs   Ethernet6/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet48    33,34     200G   9100     rs   Ethernet7/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet50    35,36     200G   9100     rs   Ethernet7/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet52    37,38     200G   9100     rs   Ethernet7/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet54    39,40     200G   9100     rs   Ethernet7/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet56    49,50     200G   9100     rs   Ethernet8/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet58    51,52     200G   9100     rs   Ethernet8/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet60    53,54     200G   9100     rs   Ethernet8/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet62    55,56     200G   9100     rs   Ethernet8/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet64    89,90     200G   9100     rs   Ethernet9/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet66    91,92     200G   9100     rs   Ethernet9/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet68    93,94     200G   9100     rs   Ethernet9/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet70    95,96     200G   9100     rs   Ethernet9/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet72    73,74     200G   9100     rs  Ethernet10/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet74    75,76     200G   9100     rs  Ethernet10/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet76    77,78     200G   9100     rs  Ethernet10/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet78    79,80     200G   9100     rs  Ethernet10/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet80    65,66     200G   9100     rs  Ethernet11/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
 Ethernet82    67,68     200G   9100     rs  Ethernet11/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
[... cut ...]
Ethernet504  497,498     200G   9100     rs  Ethernet64/1   trunk      up       up  OSFP 8X Pluggable Transceiver         off
Ethernet506  499,500     200G   9100     rs  Ethernet64/3   trunk      up       up  OSFP 8X Pluggable Transceiver         off
Ethernet508  501,502     200G   9100     rs  Ethernet64/5   trunk      up       up  OSFP 8X Pluggable Transceiver         off
Ethernet510  503,504     200G   9100     rs  Ethernet64/7   trunk      up       up  OSFP 8X Pluggable Transceiver         off
root@FRA26-0101-0809-01BS0:~#
```

prgeor, 2024-10-23:
LED brightness is set to GREEN

```
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp1/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp2/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp3/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp8/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp10/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp12/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp16/brightness
1
root@FRA26-0101-0809-01BS0:~# cat /sys/class/leds/osfp42/brightness
1
root@FRA26-0101-0809-01BS0:~#
```

prgeor, 2024-10-23:
```
admin@FRA26-0101-0809-01BS0:~$ show ver

SONiC Software Version: SONiC.20231110.19
SONiC OS Version: 11
Distribution: Debian 11.10
Kernel: 5.10.0-23-2-amd64
Build commit: 29d024f4c5
Build date: Wed Aug  7 08:14:09 UTC 2024
Built by: azureuser@cfd1eb31c000000

Platform: x86_64-arista_7060x6_64pe
HwSKU: Arista-7060X6-64PE-256x200G
ASIC: broadcom
ASIC Count: 1
Serial Number: HBG242101WL
Model Number: DCS-7060X6-64PE
Hardware Revision: 02.00
Uptime: 22:18:32 up 36 days,  4:35,  1 user,  load average: 3.52, 3.53, 3.26
Date: Wed 23 Oct 2024 22:18:32

Docker images:
REPOSITORY                 TAG           IMAGE ID       SIZE
docker-dhcp-server         latest        790325f22fef   580MB
docker-sonic-telemetry     20231110.19   55caed5f0d53   369MB
docker-sonic-telemetry     latest        55caed5f0d53   369MB
docker-mux                 20231110.19   8e64ebd400c0   349MB
docker-mux                 latest        8e64ebd400c0   349MB
docker-macsec              latest        49a6eba78eab   329MB
docker-gbsyncd-broncos     20231110.19   3884ddf3fd73   352MB
docker-gbsyncd-broncos     latest        3884ddf3fd73   352MB
docker-eventd              20231110.19   057b71228c03   301MB
docker-eventd              latest        057b71228c03   301MB
docker-gbsyncd-credo       20231110.19   65d6b095d8a8   325MB
docker-gbsyncd-credo       latest        65d6b095d8a8   325MB
docker-sonic-gnmi          20231110.19   e67994a633e4   369MB
docker-sonic-gnmi          latest        e67994a633e4   369MB
docker-teamd               20231110.19   12efb59b29bf   327MB
docker-teamd               latest        12efb59b29bf   327MB
docker-orchagent           20231110.19   e2a45c1b6825   339MB
docker-orchagent           latest        e2a45c1b6825   339MB
docker-fpm-frr             20231110.19   8ad183de31ce   352MB
docker-fpm-frr             latest        8ad183de31ce   352MB
docker-dhcp-relay          latest        68a4854d83d5   311MB
docker-snmp                20231110.19   8a46c34def4c   340MB
docker-snmp                latest        8a46c34def4c   340MB
docker-syncd-brcm          20231110.19   1f7c3adffaeb   715MB
docker-syncd-brcm          latest        1f7c3adffaeb   715MB
docker-platform-monitor    20231110.19   5a24ea90c058   421MB
docker-platform-monitor    latest        5a24ea90c058   421MB
docker-router-advertiser   20231110.19   27a5a769f61c   301MB
docker-router-advertiser   latest        27a5a769f61c   301MB
docker-sonic-restapi       20231110.19   ca88c2a22c55   319MB
docker-sonic-restapi       latest        ca88c2a22c55   319MB
docker-lldp                20231110.19   8db56acf4e6a   343MB
docker-lldp                latest        8db56acf4e6a   343MB
docker-database            20231110.19   777b93917148   301MB
docker-database            latest        777b93917148   301MB
docker-acms                20231110.19   87b256218bdf   348MB
docker-acms                latest        87b256218bdf   348MB
k8s.gcr.io/pause           3.5           ed210e3e4a5b   683kB

admin@FRA26-0101-0809-01BS0:~$
```

prgeor, 2024-10-24:
@bobbymcgonigle can you provide the ETA on the fix?

Staphylo, 2025-06-16:
This issue was solved by https://github.com/aristanetworks/sonic/commit/ac043558b64d994b0fca9ffb307b48a9d34ad644

closed 2025-06-16 by Staphylo