aristanetworks/sonic #105: [chassis] Image upgrade on the 7808 chassis wipes out the configuration on the Linecards

arlakshm, 2024-11-02:
The 7808 chassis in the lab is being upgrading using the command

`./testbed-cli.sh install-image vms26-t2-7800-1 str2 http://10.201.148.43/pipelines/Networking-acs-buildimage-Official/broadcom/internal-202405/sonic-aboot-broadcom-dnx.swi`

`vms26-t2-7800-1` is the T2 testbed with one supervisor and 3 Linecards (2 Clearwater2 cards and one wolverine card)

`str2` is the inventory file name 

This command upgrades the supervisor and all the linecards and waits for Linecards and supervisor to be up.

This commands errors out because the chassis is not accessible after 15 mins

On logging in from the serial port, I see these errors on the screen for long time
```
[  930.802723] pcieport 0000:06:05.0: DPC: unmasked uncorrectable error detected
[  930.802798] pcieport 0000:06:05.0: PCIe Bus Error: severity=Uncorrected (Fatal), type=Transaction Layer, (Receiver ID)
[  930.802800] pcieport 0000:06:05.0:   device [11f8:8533] error status/mask=00000020/04400000
[  930.802802] pcieport 0000:06:05.0:    [ 5] SDES
[  930.802951] pcieport 0000:06:05.0: AER: device recovery successful
[  930.853928] pcieport 0000:06:03.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[  930.953965] pcieport 0000:06:03.0:    [ 5] SDES
[  930.954113] pcieport 0000:06:03.0: AER: device recovery successful
[  931.026989] pcieport 0000:06:03.0:   device [11f8:8533] error status/mask=00000001/0000e000
[  931.432374] pcieport 0000:06:03.0: pciehp: Slot(4): No link
[  931.456587] pcieport 0000:06:03.0:    [ 0] RxErr                  (First)
[  931.556821] pcieport 0000:06:03.0: DPC: containment event, status:0x0001 source:0x0000
[  931.629785] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:05.0
[  931.757968] pcieport 0000:06:03.0: DPC: unmasked uncorrectable error detected
[  931.758044] pcieport 0000:06:03.0: PCIe Bus Error: severity=Uncorrected (Fatal), type=Transaction Layer, (Receiver ID)
[  931.892045] pcieport 0000:06:05.0: DPC: containment event, status:0x0001 source:0x0000
[  931.931107] pcieport 0000:06:03.0:   device [11f8:8533] error status/mask=00000020/04400000
[  931.931109] pcieport 0000:06:03.0:    [ 5] SDES
[  931.931303] pcieport 0000:06:03.0: AER: device recovery successful
[  932.014670] pcieport 0000:06:05.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[  932.014673] pcieport 0000:06:05.0:   device [11f8:8533] error status/mask=00000001/0000e000
[  932.014676] pcieport 0000:06:05.0:    [ 0] RxErr                  (First)
[  932.014740] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:03.0
[  932.014791] pcieport 0000:00:03.0: AER: Multiple Corrected error message received from 0000:06:05.0
[  932.015414] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:03.0
[  932.015465] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:05.0
[  932.015516] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:03.0
[  932.015567] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:05.0
[  932.059440] pcieport 0000:06:05.0: DPC: unmasked uncorrectable error detected
[  932.584515] pcieport 0000:00:03.0: AER: Corrected error message received from 0000:06:03.0
[  932.584758] pcieport 0000:06:03.0: DPC: containment event, status:0x0001 source:0x0000
[  932.584760] pcieport 0000:06:03.0: DPC: unmasked uncorrectable error detected
[  932.584836] pcieport 0000:06:03.0: PCIe Bus Error: severity=Uncorrected (Fatal), type=Transaction Layer, (Receiver ID)
[  932.584838] pcieport 0000:06:03.0:   device [11f8:8533] error status/mask=00000020/04400000
[  932.584841] pcieport 0000:06:03.0:    [ 5] SDES
[  932.584990] pcieport 0000:06:03.0: AER: device recovery successful
[  932.621666] pcieport 0000:06:05.0: PCIe Bus Error: severity=Uncorrected (Fatal), type=Transaction Layer, (Receiver ID)
[  932.749805] pcieport 0000:06:03.0: PCIe Bus Error: severity=Corrected, type=Physical Layer, (Receiver ID)
[  932.849887] pcieport 0000:06:05.0:   device [11f8:8533] error status/mask=00040020/04400000
[  932.849891] pcieport 0000:06:05.0:    [ 5] SDES
[  932.849893] pcieport 0000:06:05.0:    [18] MalfTLP                (First)
[  932.922960] pcieport 0000:06:03.0:   device [11f8:8533] error status/mask=00000001/0000e000
```

Eventually these backtraces are seen supervisor bootups

```
[  939.096369] INFO: task systemd:1 blocked for more than 311 seconds.
[  939.096373]       Tainted: G           OE      6.1.0-22-2-amd64 #1 Debian 6.1.94-1
[  939.096375] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  939.096376] task:systemd         state:D stack:0     pid:1     ppid:0      flags:0x00000002
[  939.096380] Call Trace:
[  939.096382]  <TASK>
[  939.096385]  __schedule+0x34d/0x9e0
[  939.096392]  schedule+0x5a/0xd0
[  939.096394]  schedule_timeout+0x118/0x150
[... 336 lines cut ...]
dhcp_relay_reconcile   old_config
dhcp_server_reconcile  remote_ctr.config.json
admin@sonic:~$ show vers

SONiC Software Version: SONiC.20240531.10
SONiC OS Version: 12
Distribution: Debian 12.6
Kernel: 6.1.0-22-2-amd64
Build commit: cf7d8848af
Build date: Fri Nov  1 09:34:37 UTC 2024
Built by: azureuser@53ad391cc000000

Platform: x86_64-arista_7800r3a_36dm2_lc
HwSKU: None
ASIC: broadcom
ASIC Count: 2
Serial Number: SGD21190878
Model Number: 7800R3A-36DM2-LC
Hardware Revision: 2a.00
Uptime: 02:16:07 up 5 min,  1 user,  load average: 0.63, 1.32, 0.71
Date: Sat 02 Nov 2024 02:16:07

```

Looks like the linecards are provisioned again.

arlakshm, 2024-11-02:
add arista_logs
[arista_logs.tar.gz](https://github.com/user-attachments/files/17605301/arista_logs.tar.gz)

arlakshm, 2024-11-02:
add 
[dmesg.log](https://github.com/user-attachments/files/17605307/dmesg.log)

kenneth-arista, 2024-12-09:
As discussed offline the clearing of configuration files may a consequence of the PCIe errors seen on this particular supervisor.

arlakshm, 2024-12-09:
@kenneth-arista, can you please provide a root-cause why the linecard went to provision mode.

rlhui, 2024-12-12:
@kenneth-arista  - it seems due to this issue, this testbed in our lab cannot be used.  What should be the fix for this?

kenneth-arista, 2025-01-24:
Closing for now as the issue has not been seen recently. Will collect logs if it happens again.

closed 2025-01-24 by kenneth-arista