aristanetworks/sonic #107: [chassis] Converting a replaced wolverine linecard does not enable macsec

arlakshm, 2024-11-06:
I used the script `chassis-eos2sonic.py` I followed the following steps

- power-off the linecard

- copy the image, minigraph and firmware packages in the /tmp/provision-artifacts on the suprrvisor

 
```
admin@str3-7808-sup-1:~$ ls /tmp/provision-artifacts/
arista-fwpkg-dpe-good-7800R3AK-36DM2-LC.tar.gz  minigraph-lc04.xml  sonic.swi
```
 

- I start the script to convert lc4 with this command. 

 
 `sudo /host/chassis-eos2sonic.py install-local -l 4`
 

- I can see that the firmware packages are getting downloaded`

```
 
Shell> fs0:
FS0:\> Aboot-k.efi initrd=\Aboot-i.nsh console=ttyS0,9600
POSTCODE=<A6>
POSTCODE=<A7>
[    4.415459] Running e2fsck on: /mnt/flash
[    5.031363] e2fsck on /mnt/flash took 0s

Aboot 10.0.2-26144132
Loading AUF data

Provision mode is 1, attempting static provisionning...
Configuring management network...
ma1.42 IP = 127.100.4.1/16, GW = 127.100.1.1, MAC = cc:1a:a3:f2:2b:2f
Downloading Manifest file (http://127.100.1.1:12321/provision/manifest?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f)...
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
tmp.fMMreM           100% |********************************|  1078  0:00:00 ETA
Wiping /mnt/flash...
Dowloading http://127.100.1.1:12321/provision/minigraph.xml?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/minigraph.xml
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
minigraph.xml        100% |********************************|  187k  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/sonic.swi?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/sonic.swi
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
sonic.swi            100% |********************************|  996M  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC.tar.gz?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC.tar.gz
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
arista-fwpkg-dpe-goo 100% |********************************|  394k  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/.provision?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/.provision
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
Dowloading http://127.100.1.1:12321/provision/do-not-clean?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/do-not-clean
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
Dowloading http://127.100.1.1:12321/provision/boot-config?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/boot-config
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
boot-config          100% |********************************|    20  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/wolverine_scd_02_06_dpe.abit?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/wolverine_scd_02_06_dpe.abit
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
wolverine_scd_02_06_ 100% |********************************| 1313k  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/platform_components.json?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/platform_components.json
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
platform_components. 100% |********************************|   459  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/metadata.json?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/arista-fwpkg-dpe-good-7800R3AK-36DM2-LC/metadata.json
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
metadata.json        100% |********************************|    86  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/firmware-update/manifest.json?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/firmware-update/manifest.json
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
manifest.json        100% |********************************|   143  0:00:00 ETA
Dowloading http://127.100.1.1:12321/provision/firmware-update/clearwater2_scd_10_15.abit?slotId=4&serial=SGD232207JY&mac=cc:1a:a3:f2:2b:2f to /mnt/flash/firmware-update/clearwater2_scd_10_15.abit
Connecting to 127.100.1.1:12321 (127.100.1.1:12321)
[... 111 lines cut ...]
2024 Nov  6 00:09:47.558393 str3-7800-lc4-1 INFO syncd1#supervisord: syncd 0:dbal_hl_register_set:  Error 'Operation timed out' indicated ; #015#015
```
 
I see this on the wolverine LC
 
```
admin@str3-7800-lc4-1:~$ cat /host/firmware-update/manifest.json

{

    "Scd(addr=0000:08:00.0)": {

        "firmware": "/host/firmware-update/wolverine_scd_02_06_dpe.abit",

        "status": "pending"

    }

}admin@str3-7800-lc4-1:~$
```
```
admin@str3-7800-lc4-1:~$ ls /host/firmware-update/
clearwater2_scd_10_15.abit  manifest.json  wolverine_scd_02_06_dpe.abit
admin@str3-7800-lc4-1:~$
```

arlakshm, 2024-11-06:
After installing the packages manually using `arista-fwutil` the macsec engine is enabled and LC comes up. 

```
admin@str3-7800-lc4-1:~$ sudo arista-fwutil updateINFO: Scd(addr=0000:08:00.0): current version 0x20001INFO: Scd(addr=0000:08:00.0): upgrading to version 0x20106[1] Sanitizing[1] Sanitizing done[1] Chip Identification: mfr=0x000000c8, memType=00000040, memCapacity=00000015[1] Chip info: numSectors=32,numPagesPerSector=256,numBytesPerPage=256[1] FileSize=1344894 bytes[1] Reading: startSector:0, opSectors:32[2] Value read from address 0x1ffff8 is : 0xde[2] Value read from address 0x1ffff9 is : 0xad[2] Value read from address 0x1ffffa is : 0xbe[2] Value read from address 0x1ffffb is : 0xef[2] Value read from address 0x1ffffc is : 0x0[2] Value read from address 0x1ffffd is : 0x0[2] Value read from address 0x1ffffe is : 0x0[2] Value read from address 0x1fffff is : 0x15[1] MagicNumber: 0xdeadbeef[1] NumSectorsToErase: 0x00000015[1] Erasing 21 sectors[1] Starting Page Programming[1] Writing 21 in last four bytes[1] startSector:0, opSectors:32[1] Programming done.INFO: Scd(addr=0000:08:00.0): upgraded to version 0x20106INFO: Upgrade requires a powercycle, doing it now
```

kenneth-arista, 2024-11-26:
Closing since we understand `arista-fwutil` is needed for this workflow to succeed.

closed 2024-11-26 by kenneth-arista