aristanetworks/sonic #110: [chassis] thermal overload seen on LC

liamkearney-msft, 2024-11-15:
str2-7804-lc7 :

```
admin@str2-7804-lc7-1:~$ show reboot-cause hist
Name                 Cause                                                                                                      Time                             User    Comment
-------------------  ---------------------------------------------------------------------------------------------------------  -------------------------------  ------  ---------
2024_11_14_23_44_13  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3016, time: 2024-11-14 23:43:18)  N/A                              N/A     Unknown
2024_11_14_23_02_48  reboot                                                                                                     Thu Nov 14 10:58:45 PM UTC 2024          N/A
2024_11_14_20_59_10  Unknown                                                                                                    N/A                              N/A     N/A
2024_11_14_03_48_12  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3014, time: 2024-11-14 03:47:12)  N/A                              N/A     Unknown
2024_11_14_03_22_56  Unknown                                                                                                    N/A                              N/A     N/A
2024_11_14_02_03_46  reboot                                                                                                     Thu Nov 14 01:59:41 AM UTC 2024          N/A
2024_11_14_01_54_41  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3011, time: 2024-11-14 01:53:45)  N/A                              N/A     Unknown
2024_11_14_01_34_58  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3010, time: 2024-11-14 01:34:05)  N/A                              N/A     Unknown
2024_11_14_00_28_47  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3009, time: 2024-11-14 00:27:47)  N/A                              N/A     Unknown
2024_11_14_00_15_29  Thermal Overload: CPU (cpu-overtemp, description: detailed fault powerup=3007, time: 2024-11-13 05:21:23)  N/A                              N/A     Unknown
```

patrickmacarthur, 2024-11-19:
Do you see similar reboot cause entries on other linecards on the same chassis?

patrickmacarthur, 2024-11-20:
Also, are you intentionally rebooting the linecard or chassis at the times that you see these thermal overload entries?

liamkearney-msft, 2024-11-25:
hey @patrickmacarthur , we dont seem to see this on other linecards in the chassis.
I noticed this while running reboot tests (so yes some intentional reboots were happening), but I can't confirm whether these all line up / are directly correlated

patrickmacarthur, 2024-11-25:
Can you try running the reboot test and then see what reboot-cause entries are added to the history?

deepak-singhal0408, 2025-01-24:
uploaded techsupport from a linecard running 202405 image at this location: https://microsoft.sharepoint.com/:u:/t/AristaSONiCWorkingGroup/EamWjCbY8MVLr_d2iL3ldfwBYeNvZo41H9CVcDc9Dmwo3g?e=ni5OeD

this linecard had a few instances of reboots(kernel-panics) and in all scenarios, reboot-history shows thermal-overload.

patrickmacarthur, 2025-04-08:
We believe that these entries are a false-positive result. There will be a future platform library update that should fix these false-positive entries.