aristanetworks/sonic #114: [chassis] pcie.yml is empty on the supervisor module

arlakshm, 2024-12-05:
Hi,
On the 202405 images, the pcie.yml is empty. Please update to so that we have pcie address for all fabric asics and linecard pcie devices

```
admin@str3-7808-sup-1:/usr/share/sonic/device/x86_64-arista_7800_sup$ cat pcie.yaml
[]
admin@str3-7808-sup-1:/usr/share/sonic/device/x86_64-arista_7800_sup$
```

arlakshm, 2024-12-05:
[kenneth-arista](https://github.com/kenneth-arista) @patrickmacarthur  for viz

Staphylo, 2024-12-06:
The pcie.yaml file containing an empty array is expected for all Arista products.
This file is actually not used for any meaningful logic on our products, instead we implement the Pcie plugin.
The file is present to make some early check of `pcie-check.sh` happy, but the plugin is what's actually used.
You can run `show platform pcie` or `pcieutil check` to see that we are actually checking for pcie devices.

Hardcoding PCIe devices and addresses for a modular system is just not realistic.
A modular system can by design have different set of cards inserted in different locations.
This translates into a physical topology that change significantly from one chassis to the next.
We therefore took the approach of overriding the default Pcie plugin for our own.

Where the default plugin reads the expected pcie topology from the yaml file, our approach generate the expected PCIe topology based on which cards are present and their physical location.
For example a 4 slot chassis with a wolverine in slot 3 will yield a different topology as a 8 slot chassis with a wolverine on slot 4.
In our case, this dynamic approach also has the advantage of solving the case where the bus allocation is changed due to some BIOS change.

closed 2024-12-06 by Staphylo