aristanetworks/sonic #116: [chassis] 400g link goes down after reboot and never comes up

arlakshm, 2024-12-16:
We are seeing an issue with the latest 2020405 image, where a 400g link fails after a reboot and does not recover. However, when we load last week's image on the same linecard, the device operates correctly and remains stable after a reboot.

Details of optics
```
admin@str3-7800-lc3-1:/usr/share/sonic/device$ show vers

SONiC Software Version: SONiC.20240510.18
SONiC OS Version: 12
Distribution: Debian 12.6
Kernel: 6.1.0-22-2-amd64
Build commit: a29e50aa51
Build date: Fri Dec 13 23:23:26 UTC 2024
Built by: azureuser@f00fcd22c000000

Platform: x86_64-arista_7800r3a_36dm2_lc
HwSKU: Arista-7800R3A-36DM2-D36
ASIC: broadcom
ASIC Count: 2
Serial Number: SGD22203294
Model Number: 7800R3A-36DM2-LC
Hardware Revision: 2a.05
Uptime: 21:07:28 up 50 min,  1 user,  load average: 2.08, 2.20, 2.25
Date: Sat 14 Dec 2024 21:07:28

Docker images:
REPOSITORY                 TAG           IMAGE ID       SIZE
docker-mux                 20240510.18   d646f3fbf191   366MB
docker-mux                 latest        d646f3fbf191   366MB
docker-macsec              latest        67a06b13593b   345MB
docker-sonic-telemetry     20240510.18   86633a3ee686   380MB
docker-sonic-telemetry     latest        86633a3ee686   380MB
docker-dhcp-server         latest        338636407fd1   338MB
docker-sonic-gnmi          20240510.18   d488984c4198   380MB
docker-sonic-gnmi          latest        d488984c4198   380MB
docker-eventd              20240510.18   34566d6833fd   314MB
docker-eventd              latest        34566d6833fd   314MB
docker-gbsyncd-broncos     20240510.18   c082e85b7203   352MB
docker-gbsyncd-broncos     latest        c082e85b7203   352MB
docker-gbsyncd-credo       20240510.18   d4c1432ff740   326MB
docker-gbsyncd-credo       latest        d4c1432ff740   326MB
docker-dhcp-relay          latest        7e376aaa0f22   326MB
docker-orchagent           20240510.18   c88f7e5d87de   356MB
docker-orchagent           latest        c88f7e5d87de   356MB
docker-fpm-frr             20240510.18   c958e6ebb126   368MB
docker-fpm-frr             latest        c958e6ebb126   368MB
docker-snmp                20240510.18   223e75162bbc   354MB
docker-snmp                latest        223e75162bbc   354MB
docker-platform-monitor    20240510.18   b850018fc1b4   434MB
docker-platform-monitor    latest        b850018fc1b4   434MB
docker-teamd               20240510.18   05565e6b7aaa   343MB
docker-teamd               latest        05565e6b7aaa   343MB
docker-lldp                20240510.18   6467d5a324b1   360MB
docker-lldp                latest        6467d5a324b1   360MB
docker-acms                20240510.18   56c2a0603509   365MB
docker-acms                latest        56c2a0603509   365MB
docker-database            20240510.18   5f8c00648c37   323MB
docker-database            latest        5f8c00648c37   323MB
docker-sonic-restapi       20240510.18   5c5c3a03404b   333MB
docker-sonic-restapi       latest        5c5c3a03404b   333MB
docker-router-advertiser   20240510.18   6287ecc27e8d   315MB
docker-router-advertiser   latest        6287ecc27e8d   315MB
docker-syncd-brcm-dnx      20240510.18   93a61d588709   703MB
docker-syncd-brcm-dnx      latest        93a61d588709   703MB
k8s.gcr.io/pause           3.5           ed210e3e4a5b   683kB

admin@str3-7800-lc3-1:/usr/share/sonic/device$ show interface transceiver  presence  Ethernet176
Port         Presence
-----------  -----------
Ethernet176  Not present

[... 163 lines cut ...]
                TxBiasLowWarning  : 5.5mA
                TxPowerHighAlarm  : 6.0dBm
                TxPowerHighWarning: 5.5dBm
                TxPowerLowAlarm   : -12.007dBm
                TxPowerLowWarning : -10.0dBm
        ModuleMonitorValues:
                Temperature: 38.781C
                Vcc: 3.334Volts
        ModuleThresholdValues:
                TempHighAlarm  : 80.0C
                TempHighWarning: 70.0C
                TempLowAlarm   : -10.0C
                TempLowWarning : 0.0C
                VccHighAlarm   : 3.5Volts
                VccHighWarning : 3.465Volts
                VccLowAlarm    : 3.1Volts
                VccLowWarning  : 3.135Volts

```

We tried the following steps to recover the link but it did not help
 - reboot linecard
 - reboot chassis
 - remove the media settting json and reboot
 - change `enable_xcvrd_sff_mgr` to false and reboot.

arlakshm, 2024-12-16:
@arista-nwolfe, @kenneth-arista, @patrickmacarthur

kenneth-arista, 2025-01-24:
Resolved by https://github.com/sonic-net/sonic-buildimage/pull/20774

closed 2025-01-24 by kenneth-arista