aristanetworks/sonic #118: [Chassis] reboot-cause is not as expected when reboot through sup (normal & abnormal)

Javier-Tan, 2025-01-22:
Hi team,

When we reboot from supervisor (both normally and abnormally), we aren't getting the expected reboot-cause in `show reboot-cause / show reboot-cause history`

Normal reboot through sup:
```
Example seen on 7800 LC:
2025_01_22_00_19_24  reboot                                              Wed Jan 22 12:15:30 AM UTC 2025                N/A
Example of expected on LC:
2025_01_22_00_46_01  reboot from Supervisor                   Wed 22 Jan 2025 12:43:35 AM UTC  admin    N/A
```

Abnormal reboot through sup:
```
Example seen on 7800LC:
2025_01_22_00_33_07  Hardware - Other (gpi-2, description: gpi 2 detailed fault, time: 2025-01-22 00:31:44)                  N/A                              N/A     Unknown
Example of expected on LC:
2025_01_22_00_38_46  Heartbeat with the Supervisor card lost  Wed 22 Jan 2025 12:36:31 AM UTC  Unknown  User issued 'Heartbeat with the Supervisor card lost' command [User: Unknown, Time: Wed 22 Jan 2025 12:36:31 AM UTC]
```

Javier-Tan, 2025-01-22:
@arlakshm for vis

arista-nwolfe, 2025-01-22:
@Javier-Tan is there a sonic-mgmt test which tests this that is failing due to this difference in reboot reason?

Javier-Tan, 2025-01-22:
@arista-nwolfe These are some of the ones we're seeing fail
|                                                                                           |                                                                                                |
|-------------------------------------------------------------------------------------------|------------------------------------------------------------------------------------------------|
| bgp/test_startup_tsa_tsb_service.py::test_tsa_tsb_service_with_supervisor_cold_reboot     | Failed: Reboot cause cold did not match the trigger Reboot from Supervisor                     |
| bgp/test_startup_tsa_tsb_service.py::test_tsa_tsb_service_with_supervisor_abnormal_reboot | Failed: Reboot cause Unknown did not match the trigger Heartbeat with the Supervisor card lost |
| bgp/test_startup_tsa_tsb_service.py::test_user_init_tsb_on_sup_while_service_run_on_dut   | Failed: Reboot cause cold did not match the trigger Reboot from Supervisor                     |
| bgp/test_startup_tsa_tsb_service.py::test_tsa_tsb_service_with_tsa_on_sup                 | Failed: Reboot cause cold did not match the trigger Reboot from Supervisor                     |

arista-nwolfe, 2025-01-23:
Looking at the driver code it looks like likely the reason for the reboot cause listing `reboot` is because our Arista specific reboot script will first reboot the LCs:
https://github.com/aristanetworks/sonic/blob/526fcf699c8a439a4192154c644ed0f5b6d0070b/arista/utils/sonic_reboot.py#L22
under the hood it looks like this ends up running the `reboot` command on the LCs:
https://github.com/aristanetworks/sonic/blob/master/arista/utils/rpc/api.py#L91

It looks like if we didn't implement our own platform specific reboot handler the code would've defaulted to running `/sbin/reboot` and likely you'd have seen the heartbeat timeout.
https://github.com/sonic-net/sonic-utilities/blob/be870a6e70376a4dd37f37d6c89c8d2f78ece079/scripts/reboot#L290

Tagging @Staphylo in case he knows the history on why we try to be more graceful about the supervisor reboot instead of using the sonic default reboot.

Javier-Tan, 2025-01-23:
> Looking at the driver code it looks like likely the reason for the reboot cause listing `reboot` is because our Arista specific reboot script will first reboot the LCs:
> 
> [sonic/arista/utils/sonic_reboot.py](https://github.com/aristanetworks/sonic/blob/526fcf699c8a439a4192154c644ed0f5b6d0070b/arista/utils/sonic_reboot.py#L22)
> 
> Line 22 in [526fcf6](/aristanetworks/sonic/commit/526fcf699c8a439a4192154c644ed0f5b6d0070b)
> 
>  def powerOffCards(platform): 
> 
> under the hood it looks like this ends up running the `reboot` command on the LCs:
> https://github.com/aristanetworks/sonic/blob/master/arista/utils/rpc/api.py#L91
> It looks like if we didn't implement our own platform specific reboot handler the code would've defaulted to running `/sbin/reboot` and likely you'd have seen the heartbeat timeout. https://github.com/sonic-net/sonic-utilities/blob/be870a6e70376a4dd37f37d6c89c8d2f78ece079/scripts/reboot#L290
> 
> Tagging [@Staphylo](https://github.com/Staphylo) in case he knows the history on why we try to be more graceful about the supervisor reboot instead of using the sonic default reboot.

Thanks this makes sense, it'll be good if we could align reboot cause with typical SONiC behvaiour for testing / standardization proccesses if possible

Javier-Tan, 2025-01-24:
@arista-nwolfe does Arista have a work around for the reboot check failures?

patrickmacarthur, 2025-01-24:
> Tagging [@Staphylo](https://github.com/Staphylo) in case he knows the history on why we try to be more graceful about the supervisor reboot instead of using the sonic default reboot.

We do a graceful reboot on the linecards because we were having issues with ext4 filesystem corruption on the linecards which we believe was caused by ungraceful reboots.

Javier-Tan, 2025-01-28:
> We do a graceful reboot on the linecards because we were having issues with ext4 filesystem corruption on the linecards which we believe was caused by ungraceful reboots.

Thanks @patrickmacarthur for more clarification.

Could you advise if Arista has workarounds for the reboot causecheck in tests? (As shown in above tests failing). Otherwise we will implement a skip for reboot cause check for Arista temporarily.

As for if reboot-cause should officially be aligned again, @arlakshm can you advise?

Thanks

kenneth-arista, 2025-02-25:
As discussed today, we will target a fix on `master` against 202505.

arista-nwolfe, 2025-03-19:
Note: We'll also want to revert our the internal sonic-mgmt patch `test-kdump-skip-lc-reboot-cause.patch` introduced in https://gerrit.corp.arista.io/c/arsonic/+/424881 once this is fixed.

Along with https://github.com/sonic-net/sonic-mgmt/pull/16755

Javier-Tan, 2025-05-09:
> Note: We'll also want to revert our the internal sonic-mgmt patch `test-kdump-skip-lc-reboot-cause.patch` introduced in https://gerrit.corp.arista.io/c/arsonic/+/424881 once this is fixed.
> 
> Along with [sonic-net/sonic-mgmt#16755](https://github.com/sonic-net/sonic-mgmt/pull/16755)

@arista-nwolfe I noticed that https://github.com/sonic-net/sonic-buildimage/pull/22391/files doesn't include abnormal reboot through sup (`Heartbeat with the Supervisor card lost`). Are there any plans to fix that one as well?

For now, after that is merged, we can only partially revert https://github.com/sonic-net/sonic-mgmt/pull/16755

arista-nwolfe, 2025-07-11:
> hi [@arista-nwolfe](https://github.com/arista-nwolfe) , what's the latest on this, for normal reboot and abnormal reboot? thanks

Hi Rita,
The normal reboot handling already merged in https://github.com/sonic-net/sonic-buildimage/pull/22391
The abnormal reboot needs approval from someone from MSFT and to be merged in https://github.com/sonic-net/sonic-host-services/pull/271

kenneth-arista, 2025-07-14:
There are also some Arista platform code changes that will require a submodule bump up as well.

kenneth-arista, 2025-08-14:
Final two PRs merged. Closing this issue
https://github.com/sonic-net/sonic-host-services/pull/296
https://github.com/sonic-net/sonic-buildimage/pull/22638

closed 2025-08-14 by kenneth-arista