aristanetworks/sonic #121: sfp reset on optical xcvrs leave the port in low-power mode

arista-nwolfe, 2025-02-25:
There is a sonic-mgmt ptest which fails due to this issue `platform_tests/api/test_sfp.py::TestSfpApi::test_reset`.

But it's also reproduce manually:
1. sfputil reset <port>
2. <port stays operationally down>
3. sfputil lpmode off <port>
4. <port comes back up>

We only started to notice this recently because the test was only updated to check port status recently
https://github.com/sonic-net/sonic-mgmt/pull/16375

https://github.com/aristanetworks/sonic-qual.msft/issues/156#issuecomment-2162873963
Looking at this comment in a issue discussing low-power mode it suggests XCVRD will only take the port out of low-power mode in these scenarios:
1. Deasserting the hardware low power mode during xcvr init
2. Asserting the hardware low power mode when a xcvr is removed/denit.

Since sfp reset won't fall under either of these scenarios it makes sense that currently xcvrd doesn't take the port out of low-power mode.

mihirpat1, 2025-02-27:
@arista-nwolfe Can you please elaborate on which change specifically from the below PR causes the current issue?
https://github.com/sonic-net/sonic-mgmt/pull/16375

arista-nwolfe, 2025-06-13:
Closing issue as https://github.com/sonic-net/sonic-platform-daemons/pull/592 has merged.

closed 2025-06-13 by arista-nwolfe