aristanetworks/sonic #122: [chassis] After conversion the linecard as no configuration

arlakshm, 2025-03-02:
After configuration with the latest 202405 image, the `minigraph.xml` is not present in `/etc/sonic/` so the linecards have not configuration.

logs
```
word: 
Last login: Sat Mar  1 04:05:48 from 10.1.84.57
Output to this terminal is being recorded for diagnostic purposes.
Note that only output that is visible on the console is recorded.

DCS-7808R3A-BND(s1)>
DCS-7808R3A-BND(s1)>
DCS-7808R3A-BND(s1)>bash
% Invalid input (privileged mode required)
DCS-7808R3A-BND(s1)>en
DCS-7808R3A-BND(s1)#bash

Arista Networks EOS shell

[autotest@DCS-7808R3A-BND ~]$ ls /tmp/provision-artifacts/
arista-fwpkg-dpe-good-7800R3A-36DM2-LC.tar.gz   minigraph-lc04.xml
arista-fwpkg-dpe-good-7800R3AK-36DM2-LC.tar.gz  minigraph-lc05.xml
chassis_eos2sonic.py                            minigraph-sup00.xml
minigraph-lc03.xml                              sonic-aboot-broadcom-dnx.swi
.py aS-7808R3A-BND ~]$ sudo chmod 755 /tmp/provision-artifacts/chassis_eos
chmod: cannot access 'a': No such file or directory
s2sonic.py  S-7808R3A-BND ~]$ sudo chmod 755 /tmp/provision-artifacts/chassis_eos
[autotest@DCS-7808R3A-BND ~]$ sudo /tmp/^C                                                                                                                         totest@DCS-7808R3A-BND provision-artifacts]$ sudo 
[autotest@DCS-7808R3A-BND ~]$ cd /tmp/provision-artifacts/                        utotest@DCS-7808R3A-BND provision-artifacts]$ sudo ./chassis_eos2sonic.py installutotest@DCS-7808R3A-BND provision-artifacts]$ sud./chassis_eos2sonic.pINFO: starting chassis provisioning scriptcts]$ sudo./chassis_eos2sonic.py install utotest@DCS-7808R3A-BND provision-artifacts]$ su./chassis_eos2sonic.py install-l
INFO: identifying current supervisor-artifacts]$ s./chassis_eos2sonic.py install-lo
INFO: found DCS-7800-SUP1A in slot 1artifacts]$ ./chassis_eos2sonic.py install-lo
INFO: checking supervisor compatibility
INFO: detecting inserted linecards for provisioning
INFO: identifying linecard 3
INFO: found 7800R3A-36DM2-LC in slot 3
WARNING: no checks done for 7800R3A-36DM2-LC in slot 3
INFO: identifying linecard 4
INFO: found 7800R3AK-36DM2-LC in slot 4
WARNING: no checks done for 7800R3AK-36DM2-LC in slot 4
INFO: identifying linecard 5
INFO: found 7800R3AK-36DM2-LC in slot 5
WARNING: no checks done for 7800R3AK-36DM2-LC in slot 5
INFO: loading artifact folder /tmp/provision-artifacts
WARNING: Unknown artifact chassis_eos2sonic.py
INFO: populate provisioning folder /mnt/flash/provision
INFO: provisioning linecard 3 (7800R3A-36DM2-LC)
INFO: provisioning linecard 4 (7800R3AK-36DM2-LC)
INFO: provisioning linecard 5 (7800R3AK-36DM2-LC)
INFO: installing SONiC on supervisor
+ true
+ unzip -ql /tmp/provision-artifacts/sonic-aboot-broadcom-dnx.swi
+ swi_installed=0
+ is_swi_installed /tmp/provision-artifacts/sonic-aboot-broadcom-dnx.swi
+ local swi_path=/tmp/provision-artifacts/sonic-aboot-broadcom-dnx.swi
++ unzip -qp /tmp/provision-artifacts/sonic-aboot-broadcom-dnx.swi .imagehash
+ local swi_version=20240532.16
++ cat /mnt/flash/image-20240532.16/.imagehash
++ :
+ local local_version=
+ '[' -z '' ']'
+ return 1
+ swi_installed=1
+ '[' 1 -ne 0 ']'
+ '[' 1 -eq 2 ']'
+ '[' -n '' ']'
+ '[' 1 -ne 3 ']'
+ false
+ false
+ regular_install
+ '[' -e /mnt/flash/image-20240532.16 ']'
+ mkdir -p /mnt/flash/image-20240532.16
[... 763 lines cut ...]
 / ___| / _ \| \ | (_)/ ___|
 \___ \| | | |  \| | | |
  ___) | |_| | |\  | | |___
 |____/ \___/|_| \_|_|\____|

-- Software for Open Networking in the Cloud --

Unauthorized access and/or use are prohibited.
All access and/or use are subject to monitoring.

Help:      https://sonic-net.github.io/SONiC/
Wiki:      https://microsoft.sharepoint.com/teams/WAG/AzureNetworking/Wiki/SONiC.aspx
On-Call:   https://portal.microsofticm.com/imp/v3/oncall/current?serviceId=10045&teamIds=26162
Dashboard: https://aka.ms/sonic-dri
Contact:   sonicdev@microsoft.com

admin@sonic:~$ ls /etc/sonic/
asic_config_checksum   dhcp_server_reconcile    old_config
constants.yml          fast-reboot_order        remote_ctr.config.json
copp_cfg.json          frr                      snmp.yml
core_analyzer.rc.json  generated_services.conf  sonic_release
credentials            grpc_secrets.json        sonic_version.yml
default_users.json     init_cfg.json            swss_dependent
dhcp_relay_reconcile   macsec_reconcile         warm-reboot_order
```

arlakshm, 2025-03-02:
@patrickmacarthur @Staphylo @kenneth-arista for viz..

This issue was not seen on earlier `202405` images.

arlakshm, 2025-03-02:
@rlhui for viz..

closed 2025-04-22 by patrickmacarthur

liamkearney-msft, 2025-04-29:
hey @patrickmacarthur , where abouts can I access this new version? I can't seem to find it on the support / software downloads page.

Thanks