aristanetworks/sonic #123: Arista conversion fails because of supported sku mismatch on supervisor

arlakshm, 2025-03-03:
Hi Kenneth, Samuel,

We are piloting the conversion from EOS to SONiC on an Arista 7808 in our staging region. 
During the conversion workflow, the chassis_eos2sonic.py script failed with the following error:

[arlakshm@STG01-0101-0400-01T2 ~]$ cd /mnt/flash
[arlakshm@STG01-0101-0400-01T2 flash]$ more install.SONiC.20220532.72.log    
INFO: starting chassis provisioning script
INFO: identifying current supervisor
INFO: found DCS-7800A-SUP1A in slot 1
INFO: checking supervisor compatibility
ERROR: This supervisor DCS-7800A-SUP1A in not supported list: ['DCS-7800-SUP1A']
ERROR: The supervisor model is not supported
[arlakshm@STG01-0101-0400-01T2 flash]$       

We had manually converted for EOS to SONiC. Is this check correct in the script?  Please let me know.

Inventory details from the device
STG01-0101-0400-01T2(s1)#show inv
System information
  Model                    Description                                                                                            
  ------------------------ -------------------------------------------------------------------------------------------------------
  DCS-7808-CH              Switch Chassis. 2 Supervisor slots, 8 Linecard slots, 6 Fabric Module slots, with AC or DC Power Supply

  HW Version  Serial Number  Mfg Date   Epoch
  ----------- -------------- ---------- -----
  11.10       TMO20260377    2020-06-24 01.00

System has 16 card slots
  Module            Model                  HW Version       Serial Number       Mfg Date         Epoch
  ----------------- ---------------------- ---------------- ------------------- ---------------- -----
  Supervisor1       DCS-7800-SUP1A         12.00            SSN20120068         2020-05-01       01.00
  Supervisor2       Not Inserted                                                                      
  Linecard3         7800R3A-36DM2-LC       11.00            SGD21270368         2021-07-24       01.00
  Linecard4         7800R3-48CQ2-LC        11.03            SGD21302663         2021-09-15       01.00
  Linecard5         7800R3-48CQ2-LC        10.10            SSN20310809         2020-09-29       01.00
  Linecard6         7800R3-48CQ2-LC        10.00            SSN20150381         2020-09-22       01.00
  Linecard7         7800R3-48CQ2-LC        11.03            SGD21356421         2021-11-04       01.00
  Linecard8         Not Inserted                                                                      
  Linecard9         Not Inserted                                                                      
  Linecard10        Not Inserted                                                                      
  Fabric1           7808R3-FM2             10.00            SSN20210019         2020-06-16       01.00
  Fabric2           7808R3-FM2             10.00            SSN20290027         2020-08-18       01.00
  Fabric3           7808R3-FM2             10.00            SSN20290028         2020-08-24       01.00
  Fabric4           7808R3-FM2             10.00            SSN20290025         2020-08-21       01.00
  Fabric5           7808R3-FM2             10.00            SSN20290036         2020-08-21       01.00
  Fabric6           7808R3-FM2             10.00            SSN20210022         2020-06-16       01.00

rlhui, 2025-03-13:
@Staphylo , can we please get ETA for this one? thanks.

arista-nwolfe, 2025-03-17:
We've fixed this by adding `DCS-7800-SUP1A` as a valid SKU in the `chassis_eos2sonic.py` script.
MSFT made the same change on their end so closing this issue.

closed 2025-03-17 by arista-nwolfe