aristanetworks/sonic #124: console to linecard not working if the process is killed

arlakshm, 2025-03-10:
During some of the production operations some of our device interaction services login to the linecards via the serial port using the command `sudo picocom /dev/ttySCD{linecard-id}` where linecard-id is 3 to 10

Some times device interaction service does not exit this session cleanly it will kill this process using the command 
`pkill -f 'sudo picocom /dev/ttySCD{0}`

After this sometime the serial connection to the linecard gets stuck, no login prompt is seen

arlakshm, 2025-03-10:
@patrickmacarthur, for viz...

arista-nwolfe, 2025-03-11:
We weren't able to reproduce this after executing the following steps multiple times.
1. `sudo picocom /dev/ttySCD2`
2. `show platform summary`
3. In separate terminal on supervisor run `pkill -f 'sudo picocom /dev/ttySCD2`
4. `sudo picocom /dev/ttySCD2`

Note that it did require an extra `enter` before the login or terminal prompt would show up after connecting to the serial console.
I.E.
```
root@nfc407:~# sudo picocom /dev/ttySCD2
picocom v3.1
<snip>
Type [C-a] [C-h] to see available commands
Terminal ready
```
`enter`
```
root@nfc407:~# sudo picocom /dev/ttySCD2
picocom v3.1
<snip>
Type [C-a] [C-h] to see available commands
Terminal ready

root@nfc407-5:~#
```

arista-nwolfe, 2025-03-31:
Closing this issue as MSFT has changed their device interaction service to cleanup the serial console session cleanly.
They will reopen this issue if they see it again.

YatishSVC, 2025-05-20:
Hi @arista-nwolfe, 
We were recently seeing this happening again on Arista 7808 chassis.

[image]

YatishSVC, 2025-05-20:
Arista team is currently investigating the issue.

Logs from the device:

[44645.486037] systemd-journald[506]: Data hash table of /run/log/journal/127087c3f9926a4f78e0b8fa6815c80f/system.journal has a fill level at 75.0 (8534 of 11377 items, 6553600 file size, 767 bytes per hash table item), suggesting rotation.
[44645.486047] systemd-journald[506]: /run/log/journal/127087c3f9926a4f78e0b8fa6815c80f/system.journal: Journal header limits reached or header out-of-date, rotating.
[47210.692924] ------------[ cut here ]------------
[47210.692929] WARNING: CPU: 10 PID: 1353154 at kernel/time/hrtimer.c:1288 hrtimer_start_range_ns+0x2c8/0x370
[47210.692939] Modules linked in: nft_meta_bridge(E) 8021q(E) garp(E) mrp(E) linux_ngbde(OE) linux_knet_cb(OE) linux_bcm_knet(OE) psample(E) linux_user_bde(OE) linux_kernel_bde(OE) xt_hl(E) xt_CT(E) xt_tcpudp(E) ip6_tables(E) xt_conntrack(E) ebt_vlan(E) nft_compat(E) nf_tables(E) tmp468(OE) amax31790(OE) veth(E) pmbus(E) pmbus_core(E) i2c_mux_pca9541(E) i2c_mux(E) nf_conntrack_netlink(E) optoe(E) nf_conntrack(E) nf_defrag_ipv6(E) nf_defrag_ipv4(E) libcrc32c(E) xfrm_user(E) lm90(E) at24(E) regmap_i2c(E) scd_hwmon(OE) i2c_dev(E) eeprom(E) bridge(E) stp(E) llc(E) nvme_fabrics(E) binfmt_misc(E) intel_rapl_msr(E) intel_rapl_common(E) intel_uncore_frequency(E) intel_uncore_frequency_common(E) sb_edac(E) x86_pkg_temp_thermal(E) intel_powerclamp(E) coretemp(E) kvm_intel(E) kvm(E) irqbypass(E) ghash_clmulni_intel(E) sha512_ssse3(E) sha512_generic(E) bonding(E) sha256_ssse3(E) sha1_ssse3(E) tls(E) aesni_intel(E) crypto_simd(E) cryptd(E) rapl(E) intel_cstate(E) intel_uncore(E) evdev(E) ofpart(E)
[47210.693009]  iTCO_wdt(E) spi_nor(E) intel_pmc_bxt(E) scd(OE) pcspkr(E) iTCO_vendor_support(E) mtd(E) intel_pch_thermal(E) watchdog(E) uio(E) ioatdma(E) sg(E) button(E) nfnetlink(E) efi_pstore(E) fuse(E) drm(E) dm_mod(E) configfs(E) ip_tables(E) x_tables(E) autofs4(E) loop(E) ext4(E) crc16(E) mbcache(E) jbd2(E) crc32c_generic(E) zstd(E) zstd_compress(E) nvme(E) nvme_core(E) nls_utf8(E) nls_cp437(E) nls_ascii(E) vfat(E) fat(E) overlay(E) squashfs(E) sd_mod(E) t10_pi(E) crc64_rocksoft(E) crc64(E) crc_t10dif(E) crct10dif_generic(E) xhci_pci(E) xhci_hcd(E) ahci(E) libahci(E) ehci_pci(E) crct10dif_pclmul(E) crct10dif_common(E) libata(E) spi_intel_platform(E) ehci_hcd(E) crc32_pclmul(E) spi_intel(E) gpio_ich(E) crc32c_intel(E) i2c_i801(E) ixgbe(E) xfrm_algo(E) scsi_mod(E) i2c_smbus(E) lpc_ich(E) usbcore(E) tg3(E) scsi_common(E) dca(E) mdio_devres(E) usb_common(E) of_mdio(E) fixed_phy(E) fwnode_mdio(E) libphy(E) mdio(E)
[47210.693084] CPU: 10 PID: 1353154 Comm: picocom Tainted: G           OE      6.1.0-29-2-amd64 #1  Debian 6.1.123-1
[47210.693088] Hardware name: Intel Camelback Mountain CRB/Camelback Mountain CRB, BIOS Aboot-norcal7-7.1.4-14169220 11/09/2019
[47210.693090] RIP: 0010:hrtimer_start_range_ns+0x2c8/0x370
[47210.693094] Code: f8 4c 89 f6 48 89 df 65 48 03 05 9b 8f 2c 6b 48 39 c1 44 89 e0 0f 94 c1 83 f0 01 21 c1 0f b6 c9 e8 bd fa ff ff e9 dd fd ff ff <0f> 0b 48 83 c4 20 5b 5d 41 5c 41 5d 41 5e 41 5f c3 cc cc cc cc 65
[47210.693097] RSP: 0018:ffffb41e607c7a40 EFLAGS: 00010246
[47210.693100] RAX: 0000000000000000 RBX: ffff93aeb077d818 RCX: 0000000000000001
[47210.693102] RDX: 0000000000000000 RSI: 00000000000fe502 RDI: ffff93aeb077da18
[47210.693104] RBP: ffff93aeb077da18 R08: 0000000000000000 R09: 0000000000cad846
[47210.693106] R10: ffffb41e607c7a50 R11: ffff93bd3fcb29d8 R12: 000000000000007d
[47210.693108] R13: 0000000000000000 R14: ffff93af5b661000 R15: ffff93af8a65de60
[47210.693110] FS:  00007f5108d23740(0000) GS:ffff93bd3fc80000(0000) knlGS:0000000000000000
[47210.693112] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[47210.693115] CR2: 00007f5108ec04e0 CR3: 00000003ae00a006 CR4: 00000000003706e0
[47210.693117] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[47210.693118] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[47210.693120] Call Trace:
[47210.693123]  <TASK>
[47210.693127]  ? __warn+0x7d/0xc0
[47210.693132]  ? hrtimer_start_range_ns+0x2c8/0x370
[47210.693136]  ? report_bug+0xe2/0x150
[47210.693142]  ? handle_bug+0x41/0x70
[47210.693146]  ? exc_invalid_op+0x13/0x60
[47210.693149]  ? asm_exc_invalid_op+0x16/0x20
[47210.693154]  ? hrtimer_start_range_ns+0x2c8/0x370
[47210.693157]  ? __alloc_pages+0x1dc/0x330
[47210.693163]  scd_uart_startup+0xa5/0x120 [scd_hwmon]
[47210.693176]  uart_port_startup+0x123/0x280
[47210.693181]  uart_port_activate+0x43/0x70
[47210.693186]  tty_port_open+0x7e/0xd0
[47210.693190]  uart_open+0x1a/0x30
[47210.693193]  tty_open+0x16f/0x690
[47210.693198]  chrdev_open+0xc4/0x250
[47210.693204]  ? __unregister_chrdev+0x50/0x50
[47210.693209]  do_dentry_open+0x1e5/0x410
[47210.693213]  path_openat+0xb7d/0x1260
[47210.693218]  ? audit_filter_inodes.part.0+0x2e/0x120
[47210.693223]  ? audit_reset_context+0x232/0x300
[47210.693226]  do_filp_open+0xaf/0x160
[47210.693233]  do_sys_openat2+0xaf/0x170
[47210.693236]  __x64_sys_openat+0x6a/0xa0
[47210.693240]  do_syscall_64+0x55/0xb0
[47210.693243]  ? syscall_exit_to_user_mode+0x1e/0x40
[47210.693247]  ? do_syscall_64+0x61/0xb0
[47210.693249]  ? audit_filter_inodes.part.0+0x2e/0x120
[47210.693252]  ? audit_reset_context+0x232/0x300
[47210.693255]  ? exit_to_user_mode_prepare+0x40/0x1e0
[47210.693259]  ? syscall_exit_to_user_mode+0x1e/0x40
[47210.693262]  ? do_syscall_64+0x61/0xb0
[47210.693264]  ? exit_to_user_mode_prepare+0x40/0x1e0
[... cut ...]
[63537.119304] scd_uart_set_termios: scd 0000:9f:00.0: UART #0: set_termios
[63538.211999] scd_uart_start_tx: scd 0000:9f:00.0: UART #0: start tx
[63538.212008] scd_uart_transmit_chars: scd 0000:9f:00.0: UART #0: transmit chars pending=1
[63538.212013] scd_uart_stop_tx: scd 0000:9f:00.0: UART #0: stop tx
[63538.981802] scd_uart_start_tx: scd 0000:9f:00.0: UART #0: start tx
[63538.981813] scd_uart_transmit_chars: scd 0000:9f:00.0: UART #0: transmit chars pending=1
[63538.981817] scd_uart_stop_tx: scd 0000:9f:00.0: UART #0: stop tx

arista-nwolfe, 2025-06-10:
This issue has been fixed by https://github.com/aristanetworks/sonic/commit/01d530033826b8b38c9004c6e95dcf6d88ccdcb3

And the submodule bump up occurred here: https://github.com/sonic-net/sonic-buildimage/pull/22539

Closing issue

closed 2025-06-10 by arista-nwolfe